maxint-rd/TM16xx #5: reading data TM1620

gunaone, 2019-10-27:
hi @maxint-rd  , regarding the process of reading through the code at the MCU, give me a note what i can do : 

*i do with SPI slave, but it coming out data 
0,64,191,32,190,190,16,80,241,64,2,3,0,0,
with 0.00Kg

maxint-rd, 2019-10-27:
(For reference see the earlier posts in [this issue](https://github.com/rjbatista/tm1638-library/issues/50))

Only 14 values? It seems you missed some bytes. Perhaps your MCU can't handle the speed or perhaps the reading is interrupted?

Let's compare these values to the values you read with the logical analyzer:

digit | value | binary (LSB-first) | segments | display value
-----|--------|-------------------|------------|----------------
1 | 00 00 | 0 |none | blank
2 | 00 02 | 0 |none | blank
3 | FD 04 | 10111111  |  F ABDECP  | 0.
4 | 7D 06 | 10111110  |  F ABDEC    | 0
5 | 7D 08 | 10111110  |  F ABDEC    | 0
6 | 0A 0A | 01010000  |   not connected | n/a

Then let's see if we can recognize some of the SPI slave values by translating them to the reversed hexadecimal values:

dec	| bin		| reversed		| hex
-----|----------|-----------|------
0	| 0	    	| 0		| 00
64     | 01000000	| 00000010	| 02
191	| 10111111	| 11111101	| FD
32	| 00100000	| 00000100	| 04
190	| 10111110	| 01111101	| 7D
190	| 10111110	| 01111101	| 7D
16	| 00010000	| 00001000	| 08
80	| 01010000	| 00001010	| 0A
241	| 11110001	| 10001111	| 8F
64    | 01000000	| 00000010	| 02
2	| 00000010	| 01000000	| 40
3	| 00000011	| 11000000	| C0
0	| 0		| 0		| 00
0	| 0		| 0		| 00

If we put these hex values in a similar table as before, you can see how they compare and which ones are missing:

digit | value | binary (LSB-first) | segments | display value
-----|--------|-------------------|------------|----------------
1 | - - | missing  | | 
2 | 00 02 | 0 |none | blank
3 | FD 04 | 10111111  |  F ABDECP  | 0.
4 | 7D - | 10111110  |  F ABDEC    | 0
5 | 7D 08 | 10111110  |  F ABDEC    | 0
6 | 0A -  |  01010000 | not connected | n/a

As you can see the first bytes and sequence bytes 06 and 0A seem missing, but the rest of the data is there. After the 0A it follows with 8F which is a command (set display pulse-width 14/16th). Then the remainder 40 C0 00 00 seems to be the start of a new sequence.

For completeness: this were the command bytes you read with the logical analyzer:

byte(s) | content
--------|----------
1          |  command 0x02  (set display mode to 2, i.e. 6 digits of 8 segments)
2          |  command 0x40 (write data registers)
3-15    |  command 0xC0 + 12 data bytes (set starting address to 0x00, then write data bytes)
16        |  command 0x8F  (set display on, set pulse-width to 14/16)

So I guess your challenge appears to be to start reading at the proper time (eg. when first command appears), and to not miss any bytes. Good luck with that!

maxint-rd, 2020-05-28:
Closing issue as question seems answered. Please respond if questions remain and issue needs to be reopened.

closed 2020-05-28 by maxint-rd