maxint-rd/TM16xx #6: Compiling error TM1638 with ESP8266 Wemos D1?

PJLC, 2020-04-12:
I'm trying to use the examples with my wemos D1 and TM1638 board but I'm always get 
"sketch/TM16xxMatrixGFX.cpp:8:26: fatal error: Adafruit_GFX.h: No such file or directory
 #include <Adafruit_GFX.h>

                          ^
compilation terminated.
exit status 1
Erro ao compilar para a placa WeMos D1 R1."

Do you know what might be the problem?

maxint-rd, 2020-04-13:
Hello, part of the library is an extension of the Adafruit GFX library. As such it requires you to first install the Adafruit GFX library. See [here](https://github.com/maxint-rd/TM16xx#tm16xxmatrixgfx-class) for more info. I think you can find the Adafruit GFX library in the Arduino IDE library manager.

If the most recent version isn't compatible anymore, please try an earlier version. I've uploaded one version [here](https://github.com/maxint-rd/arduino-sketches/tree/master/Atmel/sketchbook/libraries/Adafruit_GFX_Library) to support my mxUnifiedIO library for the nokia display.

PJLC, 2020-04-13:
Thank you for your quick reply and information.
I did install Adafruit's GFX library. The Keyboard functions work perfectly and those are very good. Until now I couldn't figure out why the keyboard works so good and the display show very strange characters... 
Using other TM1638 libraries like @gavinlyonsrepo's gavinlyonsrepo makes the display working fine though. 
https://github.com/gavinlyonsrepo/TM1638plus

maxint-rd, 2020-04-14:
Hello PJLC, thank you for your swift reply. Good to read that the buttons work fine. As I understand you now succesfully compiled, but you see strange characters when using my library. 
Thank you also for pointing me to Gavin's library. As Gavin also notes, there are different modules available. Can you tell me if you are using a module or the bare chip? If it's a module, which module are you using? Gavin has some pictures on his github where he names them model 1 and model  2 and also mentions modules with switched nibbles.

It's very wel possible that your module has different connected LED segments than the ones I've tested with. Note that if you are using the QYF-TM1638 module, you need to include TM1638QYF.h instead of TM1638.h. There is also a TM1638 module with inverted positions, for which InvertedTM1638.h needs to be used. If you use any of those, you need to use classname TM1638QYF or InvertedTM1638 respectively.

I had a brief look at Gavin's library. If it works for you, then feel free to keep using it. My library has a different setup to support different chips of the TM16xx family. Additionally I included some other features such as support for the Arduino print() function. In the [readme](https://github.com/maxint-rd/TM16xx/blob/master/README.md) you can find more about the different features.

Hopefully this helps. Good luck!

PJLC, 2020-04-14:
Hello maxint-rd,

Thank you for your kind and good support.

I'm using a QYF-TM1638 module model 2 for testing before moving to a new PCB board and yes I'm aware about the different module versions on the market therefore I have checked with different libraries to check for differences and I think the module I have does not have the switched nibbles because I've checked with Gavin's library.

You have pointed out some details that I might missed. I will have a better look.

I will keep on trying also using your latest input. Hopefully I will find the issue of this board ;)
Thanks!

maxint-rd, 2020-04-15:
Hello PJLC, thank you for responfing and gor giving more details.

I have a similar QYF-TM1638 module available for testing, so if there's anything wrong with my library I may be able to reproduce your issues and fix them. (BTW: that module has some interesting wiring of the buttons, but I did get it working).

Good luck with your continued testing. I appreciate if you could let me know your findings.

maxint-rd, 2021-09-24:
FYI: today I published a fix for some exception that appeared when running the TM1638QYF driver. If your display is showing the characters in mirror image, your module may have different wiring. In that case you could try the TM1638Anode class, which is also for common anode TM1638 displays, but, shows the digits differently. Sometime in the future, I may choose to integrate the three different TM1638 classes in this library and add configuration options to support the various display variants.

For now I'll close this issue as the remaining issue is already submitted [here](https://github.com/maxint-rd/TM16xx/issues/12).

closed 2021-09-24 by maxint-rd