maxint-rd/TM16xx #7: Request: Reverse Row and Segments

tekcreations, 2020-06-14:
I am using a Common Anode 16 segment LED 4 digit, I would love to use this library.  
I have it working by sending
setSegments(Digit number, Segment) so its reverse to normal, but I can't get a "font lookup table" 
working where I can use setdisplay to string.

Is this a possible enhancement?

maxint-rd, 2020-06-14:
Hello @tekcreations ,

Which TM16xx chip are you using? Is it the TM1640? Are you using it on a commonly available display module or as stand-alone chip with your own wiring to connect the LED-display?

For the TM1638 there is a common anode module in the market called TM1638QYF. For this module the TM1638QYF.h should be used, which takes care of the reversed setup.
Since you mention 16 segments by four digits, can I presume you are using a different font to utilize that many segments to display different characters?

Perhaps you can take inspiration from the TM1638QYF.cpp code? In that implementation I've overloaded the setSegments method and used a memory bitmap prior to flipping the rows and columns when setting segments to display characters,

Although I do have plenty chips to experiment with, I currently don't have a 16-segment by 4 character display available. I did use the TM1640 chip, but only for driving LED-matrix displays. Do you have a link to AliExpress for the display that you are using? It would probably take a couple of months, but once I have them myself I could make an extension to the library.

tekcreations, 2020-06-15:
Heya thanks for the reply. I am using a TM1640 with a common anode display similar to https://www.aliexpress.com/item/4000037229179.html
I have gotten a "tiny" way..

The bare bones test code is below where i Specify the unit, this gives me a "G" on Both digits, "segment 11"

void sendascii(uint8_t n , uint8_t a, boolean d)
{
  //uint16_t font = pgm_read_word(alphafonttable + a);
  uint16_t font = 0b0000000010111101;
    if (d)
    font |= (1 << 14);

  for (int i = 15; i >= 0; --i)
  {
    if ((bool)(font & (1U << i)))
    {
      comdisp.setSegments(0b0000000000000011,i);
    }

My issue now is not sure how to "lookup" a char to get the Uint16. that pgm read I hacked from your code but doesnt seem to work. I'll have a look at the QYF and see if I can butcher that together somehow.

tekcreations, 2020-06-15:
This is the wiring.
[image]

maxint-rd, 2020-06-15:
Cool project. I think I'll also buy some to do some fun experiments. I see your display is 14+1 segments. On Ali I saw some 16+1 displays, but those exceed the TM1640 number of pins. Interesting to see how you are using the grid-lines to connect to the cathodes. I guess that each common anode is connected to one individual segment-line per character. Smart!

You seem to have accomplished already quite a bit. Well done!
I think that what you need to do to fix reading the font is:
   pgm_read_word(alphafonttable + a*2);
Since you are reading words (16-bit values) the font has two bytes per character. The address is a byte value and the offset for each character is therefor the character number times two. 

Hope this helps. Good luck!

maxint-rd, 2021-10-15:
Hello Kieron @tekcreations , it's been a while, but I finally found some time to work on the 5241BS 14+1 segment displays I ordered some time ago. Thanks again for the input and code you provided by e-mail last year. I hope you're doing well.

This week I published my take on this in the form of the TM1640Anode class and included further support for printing Ascii characters in TM16xxDisplay and the TM16xx base class. I intend to soon publish a new example that demonstrates the use of TM1640Anode combined with another module in a merged TM16xxDisplay object.