maxint-rd/TM16xx #8: include Library into the Arduino Library Manager

SkullKill, 2020-08-17:
might be a good idea to include it into the Arduino Library Manager?

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

maxint-rd, 2020-08-17:
Good suggestion. Thanks! I'm not sure when I find the time, but I will put this on my todo-list.

SkullKill, 2020-08-18:
p.s made this esp32 wall clock which uses the TM1638 in anode mode, with 10 Segment/digits. will try to get this to work with this library when the board comes and i have some time to solder all the components in. 

will post the design files for the board etc on github once done.

[image]

[image]

[image]

SkullKill, 2020-08-27:
got this to work for 10 digits, 8 segment, in common anode mode. 

pull request submitted 

https://github.com/maxint-rd/TM16xx/pull/9

TM1638 in Common Anode mode, 10 Digits, 8 Segment

for schematics of 1st digit connected to SEG1, 2nd digit connected to SEG2 of TM1638 etc....

Schemathics of Anode connection available here https://github.com/SkullKill/ESP32-Wall-Clock-PCB and https://github.com/SkullKill/ESP32-Wall-Clock-PCB/wiki

did not modify the TM1638QYF.h module because, it looks like this has an inverted connection for the digits , i.e 1st digit connected to SEG8, 2nd digit connected to SEG7 etc...

[image]

[image]

closed 2021-10-18 by maxint-rd