maxint-rd/TM16xx #12: Adafruit_GFX.h always needed

warloofer, 2021-01-30:
As far as I understand, Adafruit_GFX.h is needed for matrix display.
I'm trying to use only the segments display of the TM1620 (I have only put the #include TM1620 in the code), and the compiler is giving the error of Adafruit_GFX.h missing. 

Why is the library required if it is not used? (as far as I know...)

Thanks in advance

ev-olution, 2021-01-30:
I was getting same error with TMS1640. Downloaded this library: https://github.com/adafruit/Adafruit-GFX-Library . And now I get another error: "C:\......Adafruit-GFX-Library-master\Adafruit_GrayOLED.h:30:10: fatal error: Adafruit_I2CDevice.h: No such file or directory"

Not sure why these errors exist as these libraries are not used.

maxint-rd, 2021-01-30:
Hello, you raise a good point. Thanks!
At the moment I don't know how to dynamically skip including the part of the library that you don't need. As far as I can see the Arduino IDE first builds the library by compiling all cpp files in the folder. After that the sketch .ino is compiled.
I think a work around can be to delete the library files you don't need (TM16xxMatrixGFX.cpp and TM16xxMatrixGFX.h) or to rename them to .txt

The dependencies of the Adafruit GFX library has changed over the past years and now it indeed requires some other Adafruit support files. I think the reason is that they made their library more hardware-independent.

ev-olution, 2021-01-30:
Thanks for your library! It works to me with a trick from my previous post. And your library is super helpful for my current project. I am using it to drive WINRISE HBS640 driver. Works perfectly with your TMS1640 library.

maxint-rd, 2021-02-21:
Today I ran into a similar problem. When compiling for an ATtiny44A (using ATtinyCore), the latest version of Adafruit GFX along with Adafruit BusIO wouldn't compile due to some missing definition in the latter part. As my project didn't use any GFX functionality I removed the TM16xxMatrixGFX files to get TM16xx compiled without issue. 

At the moment I think moving TM16xxMatrixGFX into a separate library might be a good solution to reduce dependency and compilation errors on MCU's that are not supported by Adafruit GFX. I think there are plenty of people that only use this library for 7-segment displays and don't use the dot-matrix functionality.

maxint-rd, 2024-10-11:
Note to self: can we make this dependency optional?

How about using [has_include](https://gcc.gnu.org/onlinedocs/cpp/_005f_005fhas_005finclude.html)?
```
#if defined __has_include
#  if __has_include (<stdatomic.h>)
#    include <stdatomic.h>
#  endif
#endif
```

WarlockD, 2024-12-28:
To those who don't know, even if the library is compiled that code isn't put in your project unless its used.  That said this is annoying.  Its not just a problem here but many other library's that cross link.  Wish andrino system worked with new C++ modules:P

maxint-rd, 2024-12-29:
Correct. If the GFX matrix functionality isn't used, it doesn't require any flash space.

I find it very annoying too! I prefer not to split the TM16xx library into two separate libraries., but not every user has a LED matrix module and/or wants to use it with Adafruit GFX. I haven't tested the latest Arduino IDE yet to see if __has_include can be of help here.

maxint-rd, 2025-06-05:
Finally tested with latest Arduino IDE v2.3.6 . It seems to work. Published updated code just a moment ago. Intend to do some more testing at later stage and then close this issue.

maxint-rd, 2025-12-31:
Tested to be working in Wokwi too. Closing this now. Please comment mentioning OS and version of the Arduino IDE if the issue persists.

closed 2025-12-31 by maxint-rd