maxint-rd/TM16xx #13: Example for setting dots in setDisplayToString?

tekadept, 2021-02-06:
Hi is it possible you can provide an example for setting dots in setdisplay to string?
I see it takes a word for setting dots as second parameter, but I'm not sure how to work that.
Ie what do I send to make dot 3 and 5 on?

maxint-rd, 2021-02-14:
Hello @tekadept , I will add an example as soon as I can find the time. The method setDisplayToString accepts a second parameter to set the dots, just like the method setDisplayToDecNumber(), 

The example [TM16xx_setSegments](https://github.com/maxint-rd/TM16xx/blob/master/examples/TM16xx_setSegments/TM16xx_setSegments.ino) shows on line #47 how the first four dots of the display are switched on, and the last four are switched off:
```module.setDisplayToString("HALO1234", 0xF0);```
That line probably should have some comment added.
The secondary parameter is a composite of a binary values.

To compose such value you can use the hexadecimal notation such as above. You can also use binary notation:
```module.setDisplayToString("HALO1234", B11110000);```
Another alternative is to combine bit values using | and _BV() or bit() (in case your display has more than 8 digits):
```module.setDisplayToString("HALO1234", _BV(7) | _BV(6) | _BV(5) | _BV(4));```

So to conclude, for your example setting you could try this:
```module.setDisplayToString("HALO1234", _BV(5) | _BV(3));```
If the wrong dots light up, you may want to reverse the order.

BTW. the example [TM1638_TM1637ex_two_modules](https://github.com/maxint-rd/TM16xx/blob/master/examples/TM1638_TM1637ex_two_modules/TM1638_TM1637ex_two_modules.ino) uses various other methods to show some things, including the class TM16xxDisplay, which supports the familiar Arduino print() function.

maxint-rd, 2025-04-10:
The wiki now contains the [TM16xx base class function reference](https://github.com/maxint-rd/TM16xx/wiki/TM16xx-base-class-reference).

It includes an example for using the `dots` parameter of [setDisplayToString()](https://github.com/maxint-rd/TM16xx/wiki/TM16xx-base-class-reference#setdisplaytostringconst-char-string-const-word-dots0-const-byte-pos0-const-byte-font--tm16xx_font_default).

closed 2025-04-10 by maxint-rd