maxint-rd/TM16xx #15: 2 TM16xxButtons seems to crash

tekadept, 2021-08-19:
I have 2 TM1637 chips, and I can use 2 of the tm16x matrix fine displaying LEDS, and I can use 1 Buttons class fine, but the minute I add a second and upload the arduino pro micro seems to brick and no comp port even shows up I have to hard reset and reflash without the second button for it to work.
I have enough RAM/ Flash so that doesn't seem to be the issue. any ideas?

tekadept, 2021-08-19:
Hmm might be a ramthing, on compile I have 82% ram free, and it crashes, if I make it 80% free seems to work.
I have specificed 9 buttons which is what Ihave, to avoid wasted memory already hmm.

maxint-rd, 2021-08-19:
Hello @tekadept,
To me it indeed sounds like  ram issue. In other projects I've also had weird, unpredictable behaviour due to not enough available RAM for the stack.

With regards to the TM16xxButtons library I can mention that I wanted to make it work on a ATtiny44A (which has even less RAM available), but still be able to use the full range of 32 buttons. Even though my project only used three that were actually used, I wanted to be able to freely select which buttons. For that purpose I've rewritten the library. At the cost of program memory, I now am testing a version that can work almost the same while using less RAM. Instead of storing arrays with the state and timings of a max 8/32 buttons, it now uses a number of assigned slotsto  track button states/timings. Choosing fewer slots limits RAM at cost of limited tracking of simultaneous button events.

Well to keep the long story short: this new version still needs a bit of cleanup and testing before I release it, but I hope to do that within a couple of weeks (if time permits). For the same project I also needed some other changes (to improve support of the TM1638 10x8 segments), which I want to release at around the same time.

maxint-rd, 2021-08-23:
I just published the new improved version of TM16xxButtons. 
Please note that this version allows up to 32 buttons and it uses only 6 bytes of RAM to track the state of an active button. To track combined presses (as suppported for instance by TM1638, but not on TM1637/TM1650), multiple slots can be reserved.

In this version TM16XX_BUTTONS_MAXBUTTONS is no longer used and only defined for legacy reasons. To minimize memory usage you can reduce the number of slots used by changing the value of TM16XX_BUTTONS_MAXBUTTONSLOTS.

Tested on LGT8F328P and ATtiny85 with my modified TM1637 display & keys module. ATtiny84/85 should work, but unfortunately ATtiny44/45 and smaller are too small to support the TM16xxButtons library on top of a display. However, I did manage to make a functioning clock with an ATtiny44, a DS1302 RTC and a TM1650 display and keys, using the TM1650  base class and the getButtons() method.

closed 2021-08-23 by maxint-rd