maxint-rd/TM16xx #16: Initialization functions in constructors are causing undefined behaviour, crashes ESP32

KenWillmott, 2021-09-07:
When I upload TM1637 example sketches to the ESP32 Devkit V1, the program crashes with a watchdog reset. I already found the cause, and the fix. The problem is, the fix requires a big change to several example and driver files.

It's a simple issue, actually. In some of the constructors of inherited classes like TM1637, there is initialization code that uses 'micros()'. In the Arduino environment, you can not safely call system functions from a constructor, because 'init()' which sets up functions like that, does not run until just before setup(). The reason it runs on the AVR and not on the ESP,  is because the behaviour of such functions is "undefined" and so can be processor dependent.

There are two global changes that have to be applied:
1. All class constructors must be modified to remove all display initialization code.
2. All example sketches must be modified to call setupDisplay() in setup() before any other display code is run, instead of depending on the constructor to do it.

I have already forked the library and tested the change locally, it is the only way to use the ESP32 (and possibly the ESP8266, not tested yet) with this display library. I'm able and willing to write the changes, but I thought I should invite some discussion first, as more than one file needs updating.

maxint-rd, 2021-09-08:
Hello @KenWillmott , thank you for opening this issue. In the past year I mainly used Atmel processors (and the LGT8F328P). Although I have tested this library on ESP8266, it was a while ago and subsequent changes may have introduced the issue you encountered. BTW. the ESP32 has also been used, but that was even longer ago, so I don't know quite which specific display was used then. It could very well be that at that time the ESP core that I used behaved differently than what you have used now.

I think your findings are valuable and your suggestion sound like the way to go. I think I should perform some more thorough testing (especially with ESP32), but I don't know yet when I can find the time. As you mention, the suggested change involves each display specific library file as well as the examples. I always try to test any change I publish. As this would involve many display modules, you can imagine that it will take some time. Thank you for offering to help. I may need it!

I'm not quite sure how I would like to address this overhaul. My first thought is to make the changes myself and ask you to assist in testing. Do you only have that TM1637 or do you also have other TM16xx displays?
The current projects I have laying on my desk are a TM1638 10x8 led-matrix and a TM1650 4x7 segment display (both with buttons), so a good first step for me could be to reproduce your issue on an ESP8266 and on ESP32.

KenWillmott, 2021-09-08:
That's great. I agree about the testing. May I suggest, since I already performed and tested one change for the TM1637, that I update my branch and open a pull request for only that one instance? Then you could evaluate and test it on your end. For that one, here the test is based on a ESP32 DevKit V1 board (ESP-WROOM-32 IC) connected to a Chinese "DM" TM1637 4 digit with colon display.  I downloaded the latest ESP32 core from Espressif.

I think I may have overstated the scale of the changes. I did only a brief survey of the files with this issue in view, I need to spend longer, but it is my impression that only a few classes have the problem. I will make a more careful survey this week.

For testing, I have AVR based boards including Mega2560, also 32U4 based, Due, ESP32, ESP8266, and STM32F1, F3, F4, H7. For displays, I have TM1637 4 and 6 digit version modules, TM1638 8 digit with 8 switches. Also I have a rail of TM1636 DIP IC's that I haven't tested. I haven't seen any TM1650 or others on AliExpress where I mostly find modules, but I will look again. Those would take a month to reach me, though.

maxint-rd, 2021-09-08:
Hi Ken, thank you for your contributions. Go ahead to open the pull request so I can see the precise changes you suggest. I won't immediately commit your pull to the master tree without further work.

From reading your suggestions, the change would involve changes to the base class, to all chip specific inherited classes and to all examples. As the change would break existing code that doesn't call setupDisplay(), I'm thinking of using ifdefs to make it only required on platforms that require this specific behavior. I'm curious to see how the ESP32 behaves in my current setup. My ESP32 board is quite old, probably also an early revision.

I don't have a 32U4, most STM's you mention ora TM1636, so testing those would be quite interesting.
For the TM1650 I designed this [TM1650 4-digit display module](https://github.com/maxint-rd/arduino-modules/wiki/09-TM1650-4x7-segment-display) and purchased the chips via Ali. I also made some [other modules](https://github.com/maxint-rd/arduino-modules/wiki). I submitted all of them to the public domain and added links to the project pages on JLCPCB, where you can order them.

While ordering: may I also suggest the TM1640? I find that chip great fun to work with for LED matrices and there are some modules available on Ali based on the TM1640.

KenWillmott, 2021-09-08:
No need to thank me, thank you for considering my input. Today, I went back and repeated my work in order to discover the absolute minimum changes. In the process, I found another issue, already flagged in the code - concerning the implementation of min() and max(). More about that later. With new as well as existing architectures, there is a problem of evolving tool chains, especially C++ compilers. So it is better to steer away from tool chain specific accommodations, rather better to attempt to choose methods that are more universal to most tool chains.

To address the specific issue of initialization, I think the really best approach is to implement a 'begin()' method, which calls 'clearDisplay()' and 'setupDisplay()'. It's more explicit and intuitive than calling 'setupDisplay()'. Actually, many device libraries use this paradigm, so Arduino programmers are quite used to it.

 It may be convenient to perform initialization in the constructor, but then you are faced with conditional compilation as you mentioned. That may leave an opportunity for problems with future tool chains or processors. I suggest removing initialization from the constructor, and allowing the 'begin()' method to do it, for all sub-classes.

Unfortunately, in the case of the ESP32, I am faced with both the initialization problem and also the min() macro problem. So I guess there should be two pull requests, one for each. The min() macro is only used to limit the range of a brightness parameter, but has a really deep problem - the compiler has already defined min() and max() as functions, not macros. Therefore, an '#if defined' block will not work, and the compilation fails with "multiple definition" errors.

Probably, someone knows a really elegant way to fix that - I don't, but in my opinion it would not be harmful to simply use bit masking instead of the min() macro/function. It's only used in TM16xx.cpp and in TM1650.cpp, so if it's changed in those places, the issue will never come up again. That's another thing, the ESP32 compiler is choking on things in classes that the user has not even referenced - because the errors related to TM1650.cpp came up even though I only included "TM1637.h".

Today, I notated and tested all the changes locally, but I can see that they need to be cleaned up more before an upload to the forked code base. At least, it does fully work with my NTP driven clock sketch.

Considering the testing, I would like to include headings in every example sketch, defining the pins used with each different processor or board. Not only will it help keep track of what has been tested, but a user can more easily adapt the sketch to the hardware on hand, by simply commenting or uncommenting the processor/board in use. Because, by the time I've tested everything with my menagerie of boards, there will be a lot of entries there.

KenWillmott, 2021-09-11:
Update - I opened pull request #17 on this project, to allow compilation with ESP32. It is a prerequisite for any testing because nothing will compile without it. This morning, I found an interesting result, after compiling a test program for my TM1638 "LED-KEY" board. Although an instantiation of TM1637 previously crashed the ESP, the unmodified TM1638 code in this library runs without crashing. This means that the other problem, of uninitialized functions called in the constructor, is intermittent.

So, I suggest, apply the pull or download my fork, which has no changes apart from the same pull request. You can use it to test various display drivers with ESP32.

I watched some of your videos, nice work. I am also fond of the mini-matrix displays. I tried to build one myself since there are none on the market. My 16x8 worked but when placed alongside another one, there was beat interference. I never did solve that one, so I gave up. I have other custom wired displays, but none as difficult as yours.

maxint-rd, 2021-09-14:
Hi @KenWillmott , I'm sorry I didn't have time to spend on this issue yet. Hopefully I can find some time later this week to get into this. I wonder what causes the intermittent behavior. I intend to first try to see if I can reproduce it with the versions I currently have installed, and then retest with the most recent versions of Arduino IDE and ESP8266/ESP32 cores.
Besides all the testing I want to take an effort to clean up the examples. For some basic stuff I now refer to the original library examples, but this library has evolved far enough to warrant independent examples showing the basic features (and specifics).,

Good to read you liked my videos. I found the 8x8 mini-matrix displays on Ali and designed my own PCB to make 16x8 modules. The modified Brian Lough clock uses three 16x8 TM1640 mini matrix modules using my library. I have had it running continuously for the past few years and I haven't seen any interference.

For my current project I've made a custom 10x8 matrix on perfboard, driven by a TM1638. I found that green double sided 7x9 board has just the right number of holes ( (26x31) to put the leds diagonally with transformer wire on the back. Soldering that board took quite some patience, but it wasn't too difficult.

KenWillmott, 2021-09-15:
Some updates for you - I'm convinced that the ESP compiler behaviour is incorrect. A numeric literal like '7' is supposed to be represented by the smallest available type. That is an unsigned int, or "byte". Your comparison with min() is then valid, the compiler is wrong. But a workaround of some kind is needed to compile. I suppose this problem, if it is true, should be submitted as a bug report of the compiler.

I also tested the TM1638, as I told you. But, I observed display glitches every hour or so. Then I went to the data sheet. 3.3V operation is not indicated. The TM1637 specification is not perfectly clear, but lists 5V as "typical" instead of minimum or maximum. That is why I began running the modules directly from the ESP32 3.3V supply, and eliminates the need for level translators. So while it may work with the ESP, I need to try it again with level translation and running on 5V to see if it helps.

I did notice that there are no single TM1638 examples. It wasn't very hard to produce one from the TM1637+TM1638 example, though... also there is an opportunity (although perhaps not a need) to support specific commercial modules. I haven't mentioned it yet, but I have integrated some TM1637 6 digit modules from Robodyn. They have mixed up the digit locations, so I had to over-ride a method to re-arrange the digits. I can show you all that later, but let's see what you can discover.

By the way, my Arduino core is a little out of date because I use the Mint Linux distribution, I'm too lazy to attempt an install of a more recent version. But I don't think it should really make much difference - at least not as much as the Espressif code.

maxint-rd, 2021-09-19:
Hi Ken, 

Today I performed some tests on a fresh ESP8266 (model ESP-12F). I installed the current version of the TM16xx library in the latest version of the Arduino IDE (1.8.16) and the ESP8266 core from this boardmanager: 
   https://arduino.esp8266.com/stable/package_esp8266com_index.json

I tested the TM16xxDisplay_Print example with a common TM1637 4-digit LED display using DIO=GPIO0 and CLK=GPIO4. That example has no explicit call to the setupDisplay() method.
In the IDE I kept most options to their default value. It lists Espressif FW nonos-SDK 2.2,1+100 (190703).
In that setup I could not reproduce any crashes and compilation didn't fail on the min()/max(). I will do a bit more testing, but based on my earlier experience with the ESP8266 I assume that my library is still compatible with the current ESP8266 core.

I still need to test the ESP32. As I have not used that for a long while, I first need to find a board I can use and then I intend to also do a fresh install of the most recent versions.

KenWillmott, 2021-09-22:
Great. It's possible that the difference is due to the fact that my platform is Linux. That may have resulted in a different tool chain being used. I'm assuming you are using Apple OS or Windows 10. If it works there, it makes my problem very rare. Who knows, it might even disappear with subsequent releases.

It is difficult to find information about the constructor issue. But this thread does have some comments on that:
[https://arduino.stackexchange.com/questions/30233/compiles-but-doesnt-print-anything/30234#30234](url)
The TinyPICO constructor does call millis(), but only to obtain one value. Just as is the case in an ISR, it is not used to implement any delays, but only once, to capture a value. I suspect that it is fortuitous that it returns a useful value before init() is called. The postponement method you describe is interesting, I've never seen that before.

I think the most useful thing I can do while you are investigating, is to explore the STM32 series, so I will begin on that.

maxint-rd, 2021-09-22:
Hi Ken, I'm still using windows 7 (eeeks!...). I do have one laptop running windows 10, but it's not my favorite. I use linux on servers and the Raspberry, and still intend to try Linux on the desktop. So far that never got high enough attention though...

Today I tested the TM1638QYF module on the ESP32 and encountered some issues. At first my TM16xxDisplay_Print setup remained dark. That was due to the setupDisplay() method not being called in the constructor of that driver. 

Then I went on to test the buttons using the TM16xxButtons_clicks example and got some erroneous readings. After lengthy debugging I found that the ESP32 (@240MHz) ran too fast. The CLK line exceeded 1.6 MHz and that was too much. So I took a closer look at the datasheet and indeed... It lists a maximum clockspeed of 1MHz. After adding a tiny delay it worked fine and the button events were successfully recognized. I intend to repeat that test on the ESP8266 and most likely will put a suitable delay in the base class. I also want to do a bit more ESP32 testing to see if more issues arise. 

I also have some changes waiting for release to improve support for >8 segments (e.g. the TM1638 supports 10x8).
I needed that for a led-matrix project, but also have some 13 segment LED modules which I haven't tried out yet.

I'm looking forward to read your findings on the STM32 MCU in the Arduino environment. I have bought a blue-pill module some years ago, but have not tried that one either...

maxint-rd, 2021-09-24:
Hello Ken, yesterday and today I was testing the TM1638QYF module and struggled with an exception that occurred only on the ESP8266 and only in the TM1638 QYF driver. The ESP would crash at the very first call to display some segments.
After isolating the issue and installing the Exception Decoder I found out it had to do with byte alignement of the 64-bit buffer that is used to transpose the common anode segments. Today I published a fix for this issue, as well as a fix for the problem I had due to the high speed of the ESP32 when reading key presses. 

A first question: have you installed the Exception Decoder and what does it say when your ESP32 crashes? Is the stacktrace useful?

While digging into the code, I kept thinking of an easy way to implement a fix to your initialization issue (which I still can't reproduce). I noticed that the activateDisplay parameter of the constructor could be a used for an easy intermediate improvement. If that parameter is set to false, the user intends to activate the display at a later stage and no clearing or setup of the display is required. What do you think about using that parameter? Simple examples can then be kept at minimal code, with the notice that

KenWillmott, 2021-09-28:
No, I haven't installed the Exception Decoder. I am quite new to the ESP32 environment, and when things fail, I almost always go back to the source and documentation to troubleshoot (as well as implementing in program debug steps).

I considered the initialization issue again. Disregarding for a moment the importance of it and only looking at impacts... at the moment the TM1637_extended example sketch, and I think others, do not depend on the constructor to initialize the display, but do so explicitly by calling setupDisplay() and clearDisplay() in the sketch setup() or loop().

I didn't consider this before, but doesn't that mean that if legacy users followed the pattern of the example sketches, removing initialization from the constructor would actually have no effect? I think it would only create a mystery if the user started from scratch with a sketch that doesn't include those. I think it's reasonable to expect users to follow the design patterns in the example sketches - they are, after all, test sketches as well. So they can be a baseline reference if they have a problem somewhere.

It is both good and bad that you can't reproduce the initialization issue. Good because it shows that the issue has no immediate impact. Bad because it creates a false sense of security. The problem itself is not well known because 99.999% of C++ programs run in an OS that has performed all the necessary hardware and system initialization a long time before any user program is called. A big part of the embedded programming world is more C focused and has no constructors because there are no classes. I'm just trying to explain why it's not better documented and known. It's better to understand and acknowledge the mechanism by which it works, and the nature of the assertions that guarantee correct operation on any platform, than it is to observe and judge the end results experimentally. That is because, it is impossible to perform an experiment that is guaranteed to trigger the undesired behaviour in all circumstances.

For example, it's possible that many of the compilers we're using, are actually calling init() after the constructors are called. In that case, all legitimate code would function 100% perfectly in the constructor body. However, this is not guaranteed by the language specifications so it is possible (and I have read some anecdotes of actual instances) that only some, or none of the initialization code has run. The crux of the issue, is not that it works or doesn't work, it's that it is unpredictable.

While I consider it quite important, as I pointed out at the beginning of this message, the initialization inside the constructor may not actually be necessary even for legacy code patterned after the examples you've created.

I think that your idea of making activateDisplay=false is a good one, but when compared with typical example sketches, actually would not make any difference. Again, because the examples don't make the assumption of activateDisplay=true, instead they call setupDisplay() explicitly.

To summarize that point, I doubt that removing the constructor initialization would have any real impact. Doing so is a good idea because it future-proofs the code against what is a known general problem with class initialization in stand alone embedded code such as this. Thus I don't see that efforts to accommodate it somehow, are really valuable.

Regarding testing. It's my fault, really, that I haven't fully explored the problem. I don't even have a list of the exact environment, versions, etc. Also I really need to try the same thing on Windows 10. On that note, since you say you have Windows 7, it's really important that someone try it. Here, I do have some W10 machines but they got caught in a flurry of pandemic shuffling and got offline. I plan to set one of them up, both for the ESP32 testing, and also for the STM32 testing that I plan to do. To be honest, things are kind of a mess over here, and also due to time constraints, it will take me a while to set up.

Here is a quick roadmap of my intentions:
- TM1650 4 digit display is arriving soon. I've never had one before, so I will test it.
- Set up W10 machine and try to compile the failed sketch there and see if it's different (I suspect the answer is yes)
- build a 74HCT244 level converter board and test TM1638 with ESP32 and STM32 (Blue Pill first)

maxint-rd, 2021-09-28:
Hi Ken, thank you for your elaborate response, Due to lack of time I have to keep this response short.

- Thanks for your explanation. I think I follow your line of thought and see your points of attention. As my spare time is limited I like to prioritize on issues that I can reproduce and that have the highest impact. Without reproduction I cannot properly test. 
- I've tested the TM1638 without any level conversion on both ESP32 (5V and 3v3 power to module) and on ESP8266 (5V) without level converters. In the early days it was unclear, but I recall statements about ESP8266 data pins being 5v tolerant and I thought ESP32 was 5v tolerant as well. 
- I use the portable installation of the Arduino IDE, allowing me to have multiple versions on my harddisk. Making it portable is easy: just create a directory called "Portable" below the directory that has arduino.exe. After that all sketches and libraries will be installed in that directory.

(That's all for now, I have to go...)

KenWillmott, 2021-09-30:
Okay, there is getting to be a lot going on here. Should I open another topic to only focus on testing?

I just finished a successful test of the TM1637 with the STMF103C6T8 on a "Blue Pill" board. Environment:

- OS - Windows 10 Home version 21H1
- pins - 3.3V, gnd, PB8->DIO, PB9->CLK
- display: DM 6 digit TM1637 (basically Robodyn clone)
- IDE: Arduino 1.8.12
- core: stm32duino core 2.1.0

Issues:
First compile failed because AdafruitGFX.h reported missing. So I installed Adafruit GFX. Then it failed because Adafruit_I2CDevice.h not found. Installed Adafruit_BusIO to fix that and it worked.

Hmmm... actually this test sketch doesn't use any GFX functions (I think). Why is it complaining about this?

maxint-rd, 2021-09-30:
Hi Ken,

Thank you for hanging on and for giving such clear responses.
If you want you can open another issue about testing the STM environment.
With regards to the Adafruit_GFX dependency; this is a known issue that has been reported and is recorded in another topic. In short: It's only used for LED matrix modules, but the Arduino IDE isn't aware of that and first compiles the entire library.

Unless I find a better way to deal with dependencies, I'll probably separate this part and make it a separate library. In fact, I already did that in my ESP32 installation. Easiest way to continue testing is to remove the TM166xxMatrixGFX files (both .cpp and .h). The library will compile fine without them.

Final note (unrelated): while working on some TM1650 modules, I implemented a way to combine multiple modules into one. With only sleight modifications I know have the TM16xxDisplay_ScrolledPrint example running a ticker tape that starts on a 8-digit TM1638 and continues on two 4-digit TM1650's. After cleanup and adding an example I intend to publish this new version within a week or so...

KenWillmott, 2021-10-30:
I upgraded the PC to Windows 11. 
Edition	Windows 11 Home
Version	21H2
Installed on	‎2021-‎10-‎27
OS build	22000.258
Serial number	xxxxxxxxx [redacted]
Experience	Windows Feature Experience Pack 1000.22000.258.0

System:
Device name	DESKTOP-xxxxxxxxxxx [redacted]
Processor	Intel(R) Core(TM) i5-10400 CPU @ 2.90GHz   2.90 GHz
Installed RAM	12.0 GB (11.8 GB usable)
Device ID	[redacted]
Product ID	00325-81801-60960-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

I downloaded and installed a fresh copy of the library directly from this repository. I have an ESP32 DevKit V1, display is DiyMore "4 digit LED display" which uses TM1637. The test sketch is "TM1637_test_7segment_generic" which I uploaded to my fork. ESP core installed is version 2.0.0, the latest release. Chip is ESP32-D0WDQ6 (revision 1). Arduino core is 1.8.16.

The ESP32 produces this watchdog message every few seconds:
`ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
`

Then, I comment out the initialization in the TM1637 class constructor in TM1637.cpp:
`TM1637::TM1637(byte dataPin, byte clockPin, byte numDigits, boolean activateDisplay, byte intensity)
	: TM16xx(dataPin, clockPin, dataPin, TM1637_MAX_POS, numDigits, activateDisplay, intensity)
{ // NOTE: Like the TM1640, the TM1637 only has DIO and CLK. Therefor the DIO-pin is initialized as strobe in the constructor
//	clearDisplay();
//	setupDisplay(activateDisplay, intensity);
	//delay(1);		// NOTE: using delay in constructor (called prior to Setup) may hang the MCU!
}
`

After that, it uploads and runs perfectly!

maxint-rd, 2021-11-02:
Hi Ken, thanks for posting these details. From what I see the main difference is the version of the ESP32 core. I've been using this URL in the board manager:
https://arduino.esp8266.com/stable/package_esp8266com_index.json,https://dl.espressif.com/dl/package_esp32_index.json
That URL gives me version 1.0.6 as the latest stable version. Using that version I haven't been able to reproduce the issue. 

Following you mentioning you're using version 2.0, I found newer releases on https://github.com/espressif/arduino-esp32
The 2.0 version mentioned there is a development release which supports ESP32-S2 and ESP32-C3 (both I don't have)
I think I'll give that release a test as soon as I have time and expect to be able to reproduce the crashes then.
The whole reason I want to reproduce this behavior is that I want to fix it in a way that is compatible with existing code and that keeps memory footprint low for small MCU's, such as ATtiny. I also need to spend some time to update my knowledge and see what changes this new version brings. (I'm thinking of ordering one of these newer ESP32's for further testing).

maxint-rd, 2021-11-02:
FYI: I found a post of someone encountering the [same issue in the LiquidCrystal library](https://forum.arduino.cc/t/problem-with-liquidcrystal-and-latest-esp32-core/892690] using the latest ESP32 core.

maxint-rd, 2021-11-02:
Hi again,

I have good news! (sort of...)
This afternoon I installed the [2.0.0 development release](https://github.com/espressif/arduino-esp32/releases/tag/2.0.0)  in Arduino IDE 1.8.16 using this board manager URL:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
I first removed the old core installation, but kept the library and sketches as is. 

Testing your sketch with my 4-digit TM1637 module finally gave me the looping reset that you encounter:
```rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)```
I then modified the TM1637 according your post and saw that the sketch then works as it should.
My current interpretation is that the 2.0.0 implementation of delayMicroseconds() acts more like delay(), of which I already discovered years ago that it cannot be used in a constructor.

Next step is to implement a compact, compatible fix that works well for all supported MCU's. Prior to releasing that fix, I'd like to also test it on my STM32 board, but that doesn't have top priority now.

KenWillmott, 2021-11-02:
Did you highlight the wrong board manager URL? The one you posted is for the ESP8266. I assume that your comments apply to an ESP32 board manager instead. I deliberately avoided newer releases, instead downloaded the "stable" version to help rule out unknown problems. Or, am I mistaken about that? It was dumb of me to not post the URL, it's
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
I admit to some confusion about versions, there. What I did was to act like a naive user, follow the Espressif site links to download, and then chose what seemed to be the most obvious general purpose core, hoping it would be the most commonly used.

That's probably what you have, but I'll wait to see what you say. If so, it's certainly weird that you don't see the issue. My Windows PC was purchased for mostly development, and has very minimal customization and complement of applications. It's also fairly new. So I am quite surprised about your report. It's inconvenient, too, since there really isn't much hope until you can see it on your end.

Another attempt could be made if you have a Blue Pill, I seem to recall you were in the process of obtaining one. I could repeat the above test with STM32F103C8, in that case a completely different tool chain. You could then reproduce that test over there.

Generally, ESP32 is too good a device, to be missing. It's not useful for everything, but the things it does, it does brilliantly. I even tried the on chip Hall effect sensor, it works! Kind of a strange afterthought from the designers, but there it is. The BT is useful for device network configuration as well as other things. I have about 6 of them now, 2 are already installed in projects.

I've been researching other STM lines, because I have a need for non-wifi devices. I want to future-proof myself so I'm trying to guess where things will be in a few years after some shake-out of the F103 line due to obsolescence. STM promotes the G0 and G4 series for such low cost application. Some modules just came on the market last week so I ordered a bunch of G0's. My expectation is that the G0 will eventually surpass the F0 in commercial appllications, the sales will increase and the prices will become extremely low. When those arrive, I can also test them with this library.

That's an interesting post from the Arduino forum, it lends some weight to my theory. Another interesting thing is the work-around suggested by forum member "guix" about instantiating the driver object inside the setup() function. I think any such problem could be solved that way without making any changes to the library. But it's not exactly typical code. Still, it's useful to have a "hammer fix" that will always solve the problem. It's not going to make users happy, though, because the methods have to be called with pointers, the example given was:
lcd = new LiquidCrystal( ... );
lcd->begin( ... );
All the other methods would also have to be called via pointer, not very convenient.

maxint-rd, 2021-11-02:
Aha, our posts crossed. Hope you can still read my previous post. (I should have pressed save a few minutes earlier,...)
Sorry, my first board URL contained both ESP8266 and ESP32, I edited that post.

KenWillmott, 2021-11-02:
Yes, I played with STM8. IIRC there isn't an Arduino core. There is a slight problem, though not impossible to overcome. You can freely download the STM8Cube (I think, because it was a few years ago I did that). However, the compiler isn't available freely from STM, but from Cosmic. They offer it for free, but you have to jump through some administrative and technical hoops to get it installed. You get a free license to use it only on one machine.

In my opinion, the best performance/price in the sphere of small  STM modules, are the STMF030F4P6 Demo boards: https://www.aliexpress.com/item/32828346568.html
But now, many F4 boards with a Blue Pill pinout are emerging, at incredible prices compared with last year. So that is a trend to watch as well. The F4 has all kinds of incredible goodies inside.

maxint-rd, 2024-12-24:
Hello @KenWillmott , 
It's been a while. Limited spare time can really stretch things out...

Last year I spent most spare time doing stuff with the CH32. That core closely mimics the setup of STM32. As result I believe it shares behaviour when it comes to time related  functions. I found that on CH32 a constructor should never use any time function such as millis() or delay(). Digital writes are permitted.

For the TM16xx library I recently implemented implicit begin() and will cleanup all constructors. So begin() can be called in setup(), but code doesn't break when omitted.
As time permits I'm testing all chips in my collection and updating classes following this guideline. 

Next on my Todo list is to test other MCU's including the blue pill and update/add examples. I also hope to add support for othe TM16xx chips and recently added a generic class. A few chips use I2C. To support those, having a begin() was also required.

KenWillmott, 2024-12-31:
Hi, sorry for delay in response. That is a viable approach, providing explicit initialization functions which may be overlooked on platforms where they aren't critical. I'm now focused on retrocomputing, and may try to write MC68xx assembly language versions of the drivers, or at least one as needed.

maxint-rd, 2025-01-01:
Happy new year! 

I've never done any MC68xx things, but looking at the Wiki it seems definitily quite retro indeed. Interesting but perhaps challenging, although the kind of bitbanging I did in the TM16xx library isn't too complex either.

Could you upload a photo once you get it working? I intend to add some more wiki pages, including a page collecting TM16xx projects, illustrated with pictures.

Back in the days my platform was the Z80 and I only did some base level assembly and mostly Basic on ZX81 and Spectrum. Tying one of those to some TM16xx display could be interesting too.

maxint-rd, 2025-03-27:
As far as I can seen the initialization issue has been resolved. Closing this issue now.
Thank you for your participation and for your efforts.

BTW. Next to my own ESP32 testing on hardware I also added this [online simulation](https://wokwi.com/projects/421321782868516865) for the TM1637 on ESP32 using Wokwi.