maxint-rd/TM16xx #20: Request: Function to flip 7-segment display

ngmacha, 2022-05-07:
Hello, is it possible to add a fuction that flips the TM1637-based 7-segment 4-digit display, so the numbers (time) are showing as rotated 180 degrees? I'm building a timer that can be used/mounted in 2 positions, and would like to flip the display based on the value of the accelerometer's Z orientation. Thank you.

maxint-rd, 2022-05-11:
Interesting concept - I'm sure this can be implemented, perhaps by adding an alternative font or by doing some bit-magic. I should have a GY-521 MPU-6050 accelerometer somewhere in my box of modules, but I haven't played with it yet. A few years back I made an ESP8266 driven 3-wheel robot with a TM1637 display, in which such a feature could be handy.

I'm short on time so I don't know when I get to this, but I've marked this as an enhancement and I won't close it before I looked into it. In the mean while feel free to dlve into the code of the library to see if there's an easy way to handle this...

ngmacha, 2022-05-11:
Hi @maxint-rd ! I've posted same request into several other TM11637 library repos, and got a great response from @jasoncox at https://github.com/jasonacox/TM1637TinyDisplay/issues/2 As matter of fact, he just implemented a flip function yesterday and I've tested it to work perfectly for my case. You may check with him to implement the same for your library. Thank you for replying, and kind regards - Mac Ha

maxint-rd, 2022-05-11:
Cool, thank you for your swift reply and for informing me about Jason's work. I didn't know his library and it may give me some inspiration to further improve mine. Good luck with your project and have fun!

maxint-rd, 2024-11-28:
FYI: In the mean time PR #58 was submitted by @HoseanRC to do just what is requested. My plan is to merge that PR sometime soon and at later stage determine how to change it for 15-segment support and perhaps move it to the TM16xxDisplay class, in order to keep the base class small (for small processors such ATtiny),

HoseanRC, 2024-12-12:
> FYI: In the mean time PR #58 was submitted by @HoseanRC to do just what is requested. My plan is to merge that PR sometime soon and at later stage determine how to change it for 15-segment support [...]

Just noticed myself being mentioned here
The seven segment flip code is actually just a bit of [binary hack(y) math](http://graphics.stanford.edu/~seander/bithacks.html#SwappingBitsXOR) (by Sean Eron Anderson). I just swapped segment A, B and C with D, E and F in binary level, leaving G and the dot the same since they will stay the same.
I'm not sure which 15 segment standard does this library use, but I'm quite sure it can still be flipped by bit swapping (except if it's vertically asymmetrical).
Swap the correct bits, flip the whole digit array, and you get a flipped display.

HoseanRC, 2024-12-12:
Ok, I found the 14 segment standard (which i think is what you're talking about if you include the dot), and it's kinda hard, but obviously can be done

In this case, a,b,c should be swapped with d,e,f, then g1 should be swapped with g2, and h,i,j should be swapped with k,l,m, and the dot be left behind.

You can split the variable into 3 parts:
1. & 0x0707 (abc and hij)
2. & 0x3838 (def and klm)
3. & 0x40C0 (dot, g1 and g2)

Shift the first part to left by 3 bits, Shift the second part to right by 3 bits, swap g1 and g2 in the third part, then OR all the parts together.

This will swap everything, but is going to be a bit slower than the 7 segment formula. (The formulas can be split into 2 where one is for 7 segment which is faster, and second is for 14 segment which is a bit slower.

maxint-rd, 2024-12-12:
Hello @HoseanRC, thank you for contributing to this with some more thoughts and for your suggestion. I really appreciate the simplicity of the flip7 method and have been thinking of doing something similar for 14+ segment display. 

Just to inform other readers: for alphanumeric text there are various LED displays that have more segments. Where the regular 7-segment display has 7 segments (labeled A to G) and often include a dot (DP) as an eight segment; alphanumeric displays use more segments to allow more accurate depictions of all letters of the alphabet and special characters.
I've seen displays with up to 16 segments. The ones I have for testing feature the 6 regular segments forming the outside zero character, then 8 segments on the inside and a dot. The G segment is split in G1 and G2. So it has 14 segments plus a DP segment. The same display can be found on the 14-segment x 4 HT16K33 Backpack module. (I'm working on support for that module too). I've also seen displays with s single G segment and displays with split A and D segments.

Long story short: for 13+ segments there are a couple of variations. By following a certain segment standard the number of character fonts can be kept low. My intention is to keep the base TM16xx class small, to support processors such as ATtiny44A., That base class will probably be limited to 7-segment displays and I expect your flip function is small enough to be included in the base class, but I'd like to do some testing first.

Larger processors can use more complete derived classes and add supporting classes. 
For modules that have different segment order, some mapping support will be included in the future along with improved alphanumeric support. In [this work-in-progress](https://github.com/maxint-rd/TM16xx/blob/add-generic-class-to-support-(almost)-all-TM16xx-chips/src/TM16xxIC.h) you can find the setAlphaNumeric() method that implements this.

As soon as I have a few days off I intend to continue work on this, Right now a 4-digit TM1650 project is sitting idle on my desk eagerly waiting for that to happen. Thank you once more for your contributions.

maxint-rd, 2024-12-16:
PR #58 was merged. Closing this issue.

maxint-rd, 2024-12-27:
>In this case, a,b,c should be swapped with d,e,f, then g1 should be swapped with g2, and h,i,j should be swapped with k,l,m, and the dot be left behind.

Hello @HoseanRC,
Just a small intermediate update: I found some time to experiment with 14-segment flipping. I can confirm: your solution works great. The segment order is slightly different (G1-DP-G2), but the method remains the same. I used XOR-swapping for G1 and G2 and shifting for the others. I'm still trying to understand the magic of XOR juggling to see what can be simplified. But most important: it works. Thanks for your suggestion. (Hope to release something sometime soon...)