maxint-rd/TM16xx #21: TM 1628 hometheater board

harymk, 2022-06-14:
Home theatre front panel board not same wiring from the data sheet. How to change leg details.

maxint-rd, 2022-06-14:
Hi, I don't know what this Home theater front panel board is. Are you trying to reuse a salvaged component featuring the TM1628 chip? Is it a 7-segment display? Are only the digits in different order or only the segments, or both? Please provide more info.

From what I read so far my first suggestion is to modify the font in the source code (TM16xxFonts.h) or to implement a mapping function such as done in TM1640Anode.

harymk, 2022-06-14:
this is circuit
[image]

harymk, 2022-06-16:
I understood. I have other doubt. Can i chenge digit number. 

Gird1 as second digit and gird2 as first digit. Is ther any option for that. My english not good. Help me if u can.

maxint-rd, 2022-06-16:
There is no build-in support for printing in reverse order.  The easiest way is perhaps to make your own routine to print numbers and use the setDisplayDigit function to print individual digits at the required position. 

A programmer could use the TM1628 class as a base and make a derived TM1628harymk class that overwrites the setSegments() or the endChar() method, to both change the position of the digits as well as the segments. Using such class allows you to still use all thel features in the library without further changes.

The InvertedTM1638 class uses such a technique. That class is derived from the regular TM1638 class. It changes the sendChar() method to manipulate both reversed digits, as well as changed segment positions.
(It also changes button interpretation, but you may not need that).