maxint-rd/TM16xx #22: Request: TM1616

NickLplus, 2022-06-26:
Could you please add support for TM1616 (CLK, STB, DIN, 7x4), somewhat similar to TM1620.

PDF (Chinese): http://www.titanmec.com/index.php/en/project/download/id/512.html

This chip is used in Gosund SW7 dual-channel dimmer (FCC certified it this March as 2APUZSW7), and I'm trying to control it through Tasmota.

maxint-rd, 2022-06-27:
Hello @NickLplus ,

This morning I downloaded and translated the datasheet to take a closer look. From what I can see the TM1616 is a simplified variant of the regular TM16xx chips, indeed similar to the TM1620, but even simpler. As you already mentioned, it has the same control lines (DIN/CLK/STB) as TM1620, TM1638 and others, support for only 7 segments by 4 digits and like TM1620 it doesn't support button input.

Unfortunately I don't have these chips in my supply and ordering them would take several weeks or months. From what I can see, you should be able to use the TM1620 class, as long as you initialize it using 4 digits.

This is the regular constructor: 
``` TM1620(byte dataPin, byte clockPin, byte strobePin, byte numDigits=TM1620_MAX_POS, boolean activateDisplay = true, byte intensity = 7);```
The forth parameter specifies the number of digits. Using 4 will cause the chip to initialize in 7x4 mode. From what I read in the datasheet, all commands are the same as the TM1620. Although it only has one display mode, the TM1616 datasheet states that it still needs to be initialized by sending the proper mode command. The TM1620 constructor alread does this, albeit with a different default.

Could you please see if you can confirm this chip to work okay if you use the TM1620 class?
Please be so kind to test using the TM1620  class. You could use the TM16xx_setSegments example, if you modify it to use this initialisation:
```
#include <TM1620.h>
TM1620 module(8, 9, 7, 4);   // DIN=8, CLK=9, STB=7, 4 digits
```

If it works okay I can publish a TM1616 specific class, based on TM1620 with the proper default values.

NickLplus, 2022-06-27:
Thank you, I'm already working on it, setting 1620 as a reference for 1616, just as you suggested.

The way they use TM1616 in SW7 dimmer is to control two vertical bars, each bar being composed of 7 individual LEDs: https://m.media-amazon.com/images/I/61jJzf9crRL._AC_SL1500_.jpg

maxint-rd, 2022-06-27:
I found the dimmer on Ali (for about 50 euros). It sounds like an interesting device to hack. I see that the led-bars are touch sensitive and used to display the level of dimming. Does it have an ESP8266 inside? 

(BTW. Most TM16xx chips are made for 5V, but in my experiments they usually function fine at 3.3V (white and blue leds may require a bit higher voltage). I've tested many different displays on ESP8266 and they all work fine. For ESP32 there was an object initialization issue in one of the classes, for which a have a fix, but not definitive yet).

maxint-rd, 2022-07-01:
Hello @NickLplus , Can you tell me your findings? If the library works on the TM1616 with the suggested initialization, I can publish a modified version of it for this chip and add it to the readme.

maxint-rd, 2022-07-02:
Allright, no rush... In the mean while I'm considering to order a few more different TM16xx chips, that I don't have in my collection yet. I'll include the TM1616, so I can test things myself.  That will probably be a bit longer before I get those and before I have some spare time to experiment.
I'm looking forward to see your findings. Thank you for your contribution so far...

maxint-rd, 2022-08-14:
Short update: I received the TM1616 chips. When I have time I will solder one to a breakout board, do some testing and report you my findings...

maxint-rd, 2022-08-21:
Hello, today I soldered the TM1616 to a breakout board and tested it successfully on an ESP32, using the TM1620 diver class and the TM16xxDisplay_Print and TM16xxDisplay_ScrolledPrint examples.

I defined the display as suggested previously:
```
#include <TM1620.h>
TM1620 module(15, 27, 26, 4);   // DIN, CLK, STB, four digits
```

My conclusion is: YES! You can use the TM1620 class to successfully use a TM1616 chip.
The main difference between the TM1616 and the TM1620 is that TM1616 has only one display mode: 7 segments x 4 digits. 
For now I've update the readme to reflect the outcome of this test, but I might implement a TM1616 specific class in the future to properly define its properties.

maxint-rd, 2022-08-26:
Hello, today I published a TM1616 specific class as well as an English translation of the datasheet. 
Having done that I will close this issue. Thank you for your suggestion.

closed 2022-08-26 by maxint-rd