maxint-rd/TM16xx #24: Request cs1694

harymk, 2022-08-12:
Hello, am checked with tm1668 librarie. 7 segment is working fine. I need 13 segment display. Can u help me

maxint-rd, 2022-08-12:
Yes. TM1640 supports up to 16 segments. Since last update the library supports text on more segments. I've tested 15-segment some time ago.

harymk, 2022-08-13:
Thanks for quick reply. How can i use 1668  - 13 segment mode. ```
/*
  Basic library example for TM1638. Kept small to show the simplest display functionality.
  Library based on TM1638 library by Ricardo Batista, adapted by Maxint-RD MMOLE 2018.
  
  Tested to work:
      Arduino Nano using Arduino IDE 1.8.2, Nano (Old Bootloader)), 4092 bytes flash, 135 bytes RAM

  For more information see  https://github.com/maxint-rd/TM16xx
*/

#include <TM1668.h>
//TM1668 module(8, 9, 10);   // DIO=8, CLK=9, STB=7
TM1668 module(D0, D1, D2, 4,true,7,TM1668_DISPMODE_4x13);  
#include <TM16xxDisplay.h>

//TM1628 module(D2, D1, D0);   // DIO=8, CLK=9, STB=7
TM16xxDisplay display(&module, 4);    // TM16xx object, 8 digits

void setup() {
  //display.println(F("HELLO !"));
}

int nCount=0;
void loop() {
  delay(1000);
  //display.print("Count:");
  display.println("BASS");
}

```

Its showing 7 segment ![
IMG_20220813_082825.jpg](https://user-images.githubusercontent.com/40994861/184465919-0f09fd14-6f95-45e1-991b-68bb74afd2dc.jpg)

harymk, 2022-08-13:
Hello thanks for quick reply. How can i set 13 segment on tm1668 module. 

```
/*
  Basic library example for TM1638. Kept small to show the simplest display functionality.
  Library based on TM1638 library by Ricardo Batista, adapted by Maxint-RD MMOLE 2018.
  
  Tested to work:
      Arduino Nano using Arduino IDE 1.8.2, Nano (Old Bootloader)), 4092 bytes flash, 135 bytes RAM

  For more information see  https://github.com/maxint-rd/TM16xx
*/

#include <TM1668.h>
//TM1668 module(8, 9, 10);   // DIO=8, CLK=9, STB=7
TM1668 module(D0, D1, D2, 4,true,7,TM1668_DISPMODE_4x13);  
#include <TM16xxDisplay.h>

//TM1628 module(D2, D1, D0);   // DIO=8, CLK=9, STB=7
TM16xxDisplay display(&module, 4);    // TM16xx object, 8 digits

void setup() {
  //display.println(F("HELLO !"));
}

int nCount=0;
void loop() {
  delay(1000);
  //display.print("Count:");
  display.println("BASS");
}

```

harymk, 2022-08-13:
Its only showin 7 segment 

[image]

maxint-rd, 2022-08-14:
Thank you for supplying additional information. I don't have the module you are showing, so I'n not sure to what extend I can help you. But I'll give it a try.  First thing to note is that at the moment I only used a 14-segment display on the TM1640. I haven't tried to use 13 segments on the TM1668 (yet) The library may require changes to support using a 13-segment x 4-digit display on the TM1668. 

But first some questions: 
 - In the title of this issue you mention CS1694. I am not familiar with that chip. You also mention TM1668. Is the CS1694 a chip that is fully compatible with the TM1668? Why is CS1694 in the title?
 - The datasheet of the CS1694 shows that it supports up to 11 segments, not 13. What chip is on your module?
 - I could not find information when searching for Besso SB4010-D-1. It looks like you salvaged the module from a multimedia player of some kind, Do you have a link to more information? Do you have a schematic of that module?

maxint-rd, 2022-08-15:
in addition to my previous post, I suggest you to test the segments 8-13 on your module to see if they light up and to figure out their wiring. The [TM16xx_setSegmentsDebug](https://github.com/maxint-rd/TM16xx/tree/master/examples/TM16xx_setSegmentsDebug) example may be a good base for such test, but you need to modify it to call the setSegments16() method and have the segment loop go up to uint16_t  0xFFFF instead of a byte up to 0xFF.

In the meanwhile I've been looking at the code of the library. If indeed your module features a TM1668 chip, the library requires a change to support printing 13-segment Ascii characters on the TM1668, The change needed is the addition of a function that is already implemented for TM1640Anode:
```
void TM1640Anode::sendAsciiChar(byte pos, char c, boolean fDot)
{ // Method to send an Ascii character to the display.
  // This method is also called by TM16xxDisplay.print() to display characters.
  // The base class uses the default 7-segment font to find the LED pattern.
  // Derived classes for multi-segment displays or alternate layout displays can override this method.
  uint16_t uSegments= pgm_read_word(TM16XX_FONT_15SEG+(c - 32));
	setSegments16(uSegments | (fDot ? 0b10000000 : 0), pos);
}
```
By including that method, your display.print() example should already turn on the additional segments, but they may be garbled up due to different segment wiring. To resolve that TM1640Anode implements segment mapping. Since I don't have your module or common cathode 13-seg led-displays, I cannot test such feature. If you give me more information and answer the questions in my previous response, perhaps I can provide more help.

harymk, 2022-08-15:
Ok i will chech, thanks

maxint-rd, 2022-09-07:
FYI: I have ordered some 2-digit 14-segment common cathode LED displays, Once I received them I intend to do 13-segment testing which both TM1668 and TM1628 which both support a 13-segment mode. Driving 13-segments instead of 14 probably means G1 and G2 will be combined and the decimal point will be omitted.

maxint-rd, 2022-10-19:
Hello Hari, just for your information: I have received the displays and I'm currently working on implementing 13-segment support on both TM1628 and TM1668. TM1628 works and tests fine. TM1668 should be the same. I intend to publish an updated release this week or the next.

For other news: I've ordered TM1624 and TM1623 chips that support 14 segments.

harymk, 2022-11-06:
Thanks sir.

closed 2022-11-06 by harymk