maxint-rd/TM16xx #25: Request TM1629

bdomi60, 2022-08-16:
I am a TM1629 - 8*16 led : https://datasheet.lcsc.com/lcsc/2010151506_TM-Shenzhen-Titan-Micro-Elec-TM1629-LQFP44_C19759.pdf
Does your library not reference this module?
Is this a limitation or do you plan to add it one day?
Thanks.

maxint-rd, 2022-08-17:
Hello @bdomi60, thank you for submitting this request. 
My current policy is that I try to support every TM16xx chips that I can lay my hands on. For quality purposes I don't release libraries that I cannot test. At the moment I don't have the TM1629 chip in my collection yet. If the chip is easily available on AliExpress I will add it to my shoppinglist, but it may take many weeks or even months before I get one.
If you have the skills, feel free too modify a resembling TM16xx class and submit a PR. I will keep this issue open as a reminder for future support.

maxint-rd, 2024-11-28:
Hello @bdomi60,

Many months have passed since you submitted your issue. Finally I can mention some progress: In attempt to provide support for (almost) TM16xx chips I've implemented support for the TM1629 chips in a generic class in [this experimental branch](https://github.com/maxint-rd/TM16xx/tree/add-generic-class-to-support-(almost)-all-TM16xx-chips/src), to be merged at later stage after some more testing. You can read more about  that class in [this comment](https://github.com/maxint-rd/TM16xx/issues/54#issuecomment-2381387561) of issue #54.

Based on your title "TM1629 - Buttons : 8 x 4 Multi - Segments x Digit : 8 x16 - Interface : DIO/CLK/STB" I'm not quite sure which variant of the TM1629 you have . If it has a combined DIO pin instead of two DI/DO pins, it might be one of the A/B/C/D variations. Can you take a second look? Perhaps you have the TM1629D chip? I (still) don't have one in my collection, but do support 3-wire communication and added definitions for all the TM1629 chips in [TM16xxIC.h](https://github.com/maxint-rd/TM16xx/blob/add-generic-class-to-support-(almost)-all-TM16xx-chips/src/TM16xxIC.h).

If you're still interested, could you please try that generic class using a matching definition, such as ```IC_TM1629D```and let me know your findings? (Unfortunately I don't have any 4-wire communication chips yet, so those are still on my TODO-list),

maxint-rd, 2025-04-08:
This issue is assumed to be resolved by implementation of the TM16xxIC class. I will close it now. Feel free to comment with further details if it wasn't resolved. I will reopen the issue and give it more attention.

closed 2025-04-08 by maxint-rd