maxint-rd/TM16xx #26: TM1624

blackhawk2611, 2022-09-09:
Will this code work with TM1624 led driver?

maxint-rd, 2022-09-09:
Hello @blackhawk2611 , 
A quick look at the datasheet showed that then TM1624 has these features:
   - segments x digits: 11x7 - 14x4 seg x grid (common cathode)
   - interface: DIN/CLK/STB
   - buttons: n/a
Based on these features the best match of currently supported chips is TM1620, followed by TM1628 and TM1668.
Since I don't have a TM1624 in my collection, I cannot test to see which class is most compatible. I think that driving 8 segments will probably work; using more than 8 may be problematic. How many segments do you want to use? Do you have a module that uses this chip? If so, can you post a picture? 

I suggest you to test them yourself and let me know your findings. I've added this chip to my wishlist and once I have one to test with, I can add full support to the library.

maxint-rd, 2024-11-28:
Many months have passed since you submitted your issue. Finally I can mention some progress: Some time ago I've purchased some TM1624 chips. I've implement support for that chip in a generic class in [this experimental branch](https://github.com/maxint-rd/TM16xx/tree/add-generic-class-to-support-(almost)-all-TM16xx-chips/src), to be merged at later stage after some more testing. You can read more about  that class in [this comment](https://github.com/maxint-rd/TM16xx/issues/54#issuecomment-2381387561) of issue #54.

I've you're still interested, could you please try that generic class using the definition ```IC_TM1624```and let me know your findings?

maxint-rd, 2024-12-23:
**_Update_**  
Yesterday I published TM1624.h, which offers support for the TM1624 via the TM16xxIC class. I have tested the TM1624 myself on the CH32V003 in 14x4 mode with an 15-segment alphanumeric LED display.

The TM16xxIC class was merged in the main branch, and the experimental branch will be removed at a later moment.

maxint-rd, 2025-01-02:
Done.

maxint-rd, 2025-01-02:
Do

closed 2025-01-02 by maxint-rd