maxint-rd/TM16xx #28: error: '_BV' was not declared in this scope

tbillion, 2022-11-16:
C:\Users\LocalAdmin\Documents\Arduino\libraries\TM16xx_LEDs_and_Buttons\src\TM1630.cpp: In member function 'virtual uint32_t TM1630::getButtons()':
C:\Users\LocalAdmin\Documents\Arduino\libraries\TM16xx_LEDs_and_Buttons\src\TM1630.cpp:77:31: error: '_BV' was not declared in this scope
   77 |     keys_K2 |= ((((received & _BV(1)) >> 1) | ((received & _BV(4)) >> 3)) << (2 * i));   // bit 1 for K2/KS3|5|7 and bit 4 for K2/KS2|4|6|8
      |                               ^~~
C:\Users\LocalAdmin\Documents\Arduino\libraries\TM16xx_LEDs_and_Buttons\src\TM1628.cpp: In member function 'virtual uint32_t TM1628::getButtons()':
C:\Users\LocalAdmin\Documents\Arduino\libraries\TM16xx_LEDs_and_Buttons\src\TM1628.cpp:98:29: error: '_BV' was not declared in this scope
   98 |     keys_K1 |= (( (received&_BV(0))     | ((received&_BV(3))>>2)) << (2*i));   // bit 0 for K1/KS1 and bit 3 for K1/KS2
      |                             ^~~
C:\Users\LocalAdmin\Documents\Arduino\libraries\TM16xx_LEDs_and_Buttons\src\TM1637.cpp: In member function 'virtual uint32_t TM1637::getButtons()':
C:\Users\LocalAdmin\Documents\Arduino\libraries\TM16xx_LEDs_and_Buttons\src\TM1637.cpp:107:9: error: '_BV' was not declared in this scope
  107 |  return(_BV((0xFF-received)&0x0F));   // return bit set for the button that is pressed (bits 0-15)
      |         ^~~
exit status 1

maxint-rd, 2022-11-16:
Hello @tbillion ,

Thank you for reporting this issue. For what processor are you compiling? What core are you using?
I ask this so I can try to reproduce your issue, in order to fix it.

tbillion, 2022-11-17:
I am using the philhower pico core , on ide 1.8.19. your library is
important because I have a customer board that uses the 3x8 keypad feature
of the chip and your library is the only one I've found that supports this
. If I can help anyway further please let me know .

maxint-rd, 2022-11-17:
Hi @tbillion, thank you for the further details. For the library to properly compile on the Pi Pico, you should make sure you are using the latest version, to include the fix I made in september. The added definition of _BV is in the TM16xx,h header file, 
If I can find some time I'll do another test with my own Pi Pico and publish these latest changes into a new release, to make it easier for people using the Arduino environment.

maxint-rd, 2025-03-27:
Have not encountered this issue when testing for Pico. Closing this issue now. 
Feel free to comment/reopen when the issue still occurs,

closed 2025-03-27 by maxint-rd