maxint-rd/TM16xx #30: Redefine Pins

ngsjo, 2023-02-21:
Hello, I'm looking for where the pins are defined. We have a custom application for the 1620 chip and the pins are defined differently. I'm trying to figure out where in the library to redefine which segment of the display each output pin is connected to. 

For whatever reason this is how the PCB designer set it up. 
1620 PIN     Segment
2                  A
3                  F
4                  B
5                  G
6                  C
7                  DP
8                  D
9                  E

maxint-rd, 2023-02-21:
Hello Ngsjo,

At the moment there is no specific API in the library for using a modified segment order. The library kind-of expects the segments to be wired up correctly. Depending demand I may implement such API in a future version, but this won't be very soon. So you may need to make some modifications yourself, for which I'll give you some suggestions.

The pin definition is primarily hardcoded in the font (see TM16xxFonts.h). So you could change the font to get the desired result. (First focus on only the numbers in TM16XX_NUMBER_FONT).
The example TM16xx_setSegmentsDebug.ino may be handy to test the order of the segments. The example TM16xxDisplay_Print.ino is very simple and shows a counter,

A more evolved method is implemented in the TM1640Anode class. There a mapping function maps the segments in different order using these functions:
  void TM1640Anode::setSegmentMap(const byte aMap[])
  uint16_t TM1640Anode::mapSegments16(uint16_t segments)
  void TM1640Anode::setSegments16(uint16_t segments, byte position)

If you're a programmer, you could use the TM1620 class as a base and make a derived TM1620ngsjo class that overwrites the setSegments() or the sendChar() method, to change the order of the segments (or perhaps also the position of the digits). Using such class allows you to still use all the features in the library without further changes.
The InvertedTM1638 class uses such a technique. That class is derived from the regular TM1638 class. It changes the sendChar() method to manipulate both reversed digits, as well as changed segment positions.

maxint-rd, 2024-12-23:
Hello @ngsjo 
FYI: recently the TM16xxIC class was added to provide generic support for (almost) all TM16xx chips. Although I don't have the TM1620 for testing, that class does provide support for it. 
It also includes the setAlphaNumeric() method as the API to provide a segment map. Using such a map allows to specify an alternative order of the segments. The segment map is a simple array of bytes.

If you have time to give it a test, I would appreciate your feedback. 

Please let me know if you have enough information or if you need me to give you some example code (I have not yet added such an example to the library).

ngsjo, 2024-12-23:
@maxint-rd 

I no longer work at the company I was working on this for, but I may still have a use for this on some personal projects utilizing some old sample hardware from that project. I don't know if/when I would get around to that.

maxint-rd, 2024-12-24:
@ngsjo , okay,  I recognize how limited spare time has its effects.
Thank you for responding. I will close this issue now and consider it solved. Feel free to raise a new issue or reopen as needed.

maxint-rd, 2024-12-24:
-

closed 2024-12-24 by maxint-rd