maxint-rd/TM16xx #32: TM1682

TinkerWorX, 2023-02-26:
I've found a TM1682 on a piece of equipment I took apart. I've been unable to find a datasheet for it, but I did stumble on this repository.

The IC controls a display with four 7-segment displays, ten "symbols" which I assume are just LEDs and four push buttons.
[image]
It has six symbols not visible along the top. I'm counting those as well as the four signal strength bars as the ten LEDs.

How likely is it that the library can work with this IC? It has a Titan Micro symbol that matches other TM ICs. If there's a way to test, what would I need to modify, if anything?

TinkerWorX, 2023-02-26:
I've found out this is most likely the TM1652, which also explains the single data line. It matches everything I've mapped, so I'm closing this again.

closed 2023-02-26 by TinkerWorX

maxint-rd, 2023-02-27:
Hello @TinkerWorX , thank you for submitting this issue. Nice to see those pictures as an example of the chips being used in a device.

As you probably saw, the TM1652 is a bit special, having only a single data line. I recently added support in this library, as well as my translation of the datasheet, which shows the single data line on pin 16, opposite of Vdd on pin 1. This matches the pinout of your module. 

Are you going to try it with my library? If so, please let me know if you got it working. Good luck!

TinkerWorX, 2023-02-27:
Hey @maxint-rd!

Yeah, it did throw me for a bit of a loop when I started mapping out the connections, which made me confused about what I was dealing with. Didn't help that the buttons weren't connected to the IC at all, but as you probably know the TM1652 doesn't have button inputs, so those are handled separately.

I used a section of your library ported to C# to test it over a simple USB->Serial adapter I had. It worked right away. Since the display isn't just five 7-segment displays, I need to modify things since the signal strength 4 indicator is the 8th bit of the first segment. But your code is an awesome reference and made it trivial to get it initialized and the timings right.

I've attached a picture and video of it working.

[image]

https://user-images.githubusercontent.com/1730988/221532999-1031a76b-e36e-415c-8177-b07e780ff82a.mov

maxint-rd, 2023-02-27:
Well done! Thank you for this feedback and your uploads. No button-support perhaps explains the many extra pins of the connector on your module, but I probably would have used a different TM16xx chip if that robot mower was my design. Having a pin on the connector labeled SWCLK must have been extra confusing....

Having no clock line, timing is kind of critical for the TM1652 and it made this implementation somewhat different from the other TM16xx chips. Nice to see my software helped a bit and good to know that C# on a pc is fast enough to match the 19k2 bitrate. 

Have fun finding new purpose for this component!