maxint-rd/TM16xx #36: regarding interfacing of tm1640

beginner140BB, 2023-05-04:
hey there 
i am using TM1640 and controller used is esp32-s3
is there any way to interface with this ic.
from developer point of view does he/she need to develop from scratch as there are no APIs available for TM1640 with esp32-s3

whether there any other ic available to replicate this functionality as our basic requirement is to drive common anode.
platform used ESP-IDF.

There is also confusion regarding how you have implemented the interfacing as the two wire interface mentioned in datasheet is not I2C.

beginner140BB, 2023-05-05:
okay thanks for replying
currently i am trying to convert the useful API from your library to blink single 7 segment display.
i have checked on your library and i have seen that DIN and CLK setting high and low.

what if i need to print 1 on the 7 segment display where we are using common anode.

these are the different replicated API from your library which is to be implemented on ESP32-S3

what are the different steps that i should follow with the help of manual of tm1640 to implement the functionality.

```
static void TM1640_SendByte(uint8_t byte)
{
    for (int i = 0; i < 8; i++) {
        gpio_set_level(CLK_PIN, LOW);
        gpio_set_level(DIO_PIN, byte & 0x01);
        byte >>= 1;
        gpio_set_level(CLK_PIN, HIGH);
    }
}

static void TM1640_SendCommand(uint8_t command)
{
	TM1640_SendByte(command);
    gpio_set_level(CLK_PIN, LOW);
    gpio_set_level(DIO_PIN, LOW);

}

void TM1640_Init(uint8_t DIO_Pin, uint8_t CLK_Pin)
{
    gpio_pad_select_gpio(DIO_Pin);
    gpio_pad_select_gpio(CLK_Pin);
    gpio_set_direction(DIO_Pin, GPIO_MODE_OUTPUT);
    gpio_set_direction(CLK_Pin, GPIO_MODE_OUTPUT);

    // Set the initial state of the GPIO pins
    gpio_set_level(DIO_Pin, 1);
    gpio_set_level(DIO_Pin, 1);

    // Send the initialization command to the TM1640
    TM1640_SendCommand(TM1640_ADDR_FIXED);
}

void TM1640_start()
{
	gpio_set_level(CLK_PIN, LOW);
    gpio_set_level(DIO_PIN, LOW);
	//micro second delay is required

}
void TM1640_stop()
{
	gpio_set_level(CLK_PIN, HIGH);
    gpio_set_level(DIO_PIN, HIGH);
	//micro second delay is required
}
```

maxint-rd, 2023-05-05:
Unfortunately I won't be able to able to support you every step off the way or debug/test your code, but I can give you some short tips:
  - after implementing the basic functions your pasted above, you need to test them by sending data
  - use a logical analyzer to check the output levels and timing of the signals generated by your code. I use a very cheap Saleae clone. It's essential for this and I use it all the time. 
  - keep in mind that ESP32 runs on high clockspeeds, so delays are required.
  - first implement setting the individual segments of a digit, before trying to display entire characters. Using an 8x8 led matrix can be helpful
  - to display any number or other character you need to properly set the correct bits of the display memory. 
  - for common anode transposition is required. Each digit would have its own specific bit, so showing the number one would involve at least two memory addresses.

I guess you should also look into the code of the [TM1640Anode class](https://github.com/maxint-rd/TM16xx/blob/master/src/TM1640Anode.cpp). That subclass was created to support 16-segment alpha-numeric displays. It features a bitmap in memory to take care of the common anode connections and segment mapping to allow alternative segment order.

beginner140BB, 2023-05-05:
okay will check on that

[image]
what should i do if need to display 1 i.e b and c in 7 segment
as the manual shows different steps in the transmission 

what should be the address and in the data portion sending 0x01 will serve the purpose that i am looking for or is it further more complicated process or maybe i am not able to understand the table and respective address pattern.

[image]

as shown in the diagram.

beginner140BB, 2023-05-05:
from my basic understanding this api can be used for send data to 7 segment.
but whether this API alone is needed fro printing something in the display.

```
void TM16xx::sendData(byte address, byte data)
{
  sendCommand(TM16XX_CMD_DATA_FIXED);							// use fixed addressing for data
	start();
  send(TM16XX_CMD_ADDRESS | address);						// address command + address
  send(data);
  stop();
}
```

maxint-rd, 2023-05-05:
Yes, that is correct. If you study my library well, you can find how that method is called to light up the leds to show something on the display. A first test would be to do a loop that sends subsequential data to a specific address, to see how it affects the display. You can use the [setSegments example](https://github.com/maxint-rd/TM16xx/blob/master/examples/TM16xx_setSegments/TM16xx_setSegments.ino) for doing suc a test.

One note: Since you want to use common anode connections, you would tie the segments to the COM pins and each digit to a SEG pin, allowing for maximum of 16 segments and 8 digits on the TM1640. For 8 segments of a specific digit, the data would affect 8 addresses. To properly manage those, I used a memory bitmap and transformation when showing the digits. Using common anode configuration on a TM-chip intended for common cathode is possible, but requires extra programming effort. Alternatively you could use a different TM-chip, but that is not what you state in your opening post.

maxint-rd, 2023-05-05:
Additional note: sendData() is used to store led-status in the TM-chip. Prior to doing that, the library first initializes the chip by calling setupDisplay(boolean active, byte intensity);

beginner140BB, 2023-05-08:
> Yes, that is correct. If you study my library well, you can find how that method is called to light up the leds to show something on the display. A first test would be to do a loop that sends subsequential data to a specific address, to see how it affects the display. You can use the [setSegments example](https://github.com/maxint-rd/TM16xx/blob/master/examples/TM16xx_setSegments/TM16xx_setSegments.ino) for doing such a test.
> 
> One note: Since you want to use common anode connections, you would tie the segments to the COM pins and each digit to a SEG pin, allowing for maximum of 16 segments and 8 digits on the TM1640. For 8 segments of a specific digit, the data would affect 8 addresses. To properly manage those, I used a memory bitmap and transformation when showing the digits. Using common anode configuration on a TM-chip intended for common cathode is possible, but requires extra programming effort. Alternatively you could use a different TM-chip, but that is not what you state in your opening post.

okay thanks for the suggestion
as per the example setSegment and based on the connection
[image]
i am able to turn on segments.
Need to look on how to convert effectively for  esp-idf platform.

maxint-rd, 2023-05-08:
Very good. Congratulations!
If you're interested, you could follow this up with a look at the [TM1640Anode class](https://github.com/maxint-rd/TM16xx/blob/master/src/TM1640Anode.h) I mentioned before. I made this class to show alpha-numeric characters on a 16-segment common anode display, but it should also be usable for 8 segment displays.
Good luck with your further endeavors. I wish you all the best.

closed 2023-05-10 by beginner140BB

beginner140BB, 2023-05-12:
hey there, @maxint-rd 
by using basic [](https://github.com/maxint-rd/TM16xx/blob/master/src/TM16xx.cpp) API from this library i am able to convert to api based on idf.
Need to know which api should be better if i need to use multipile 7 segments or if there any suggestions on examples that i could follow from the arduino that i could follow to select multiple segments on common anode mode.

[image]

if possible do you have any idea here as we want to connect multiple segment displays.

maxint-rd, 2023-05-12:
Hi again, I'm not quite sure what your question is. Can you clarify?

Personally I like to use the [TM16xxDisplay class](https://github.com/maxint-rd/TM16xx#tm16xxdisplay-class), which allows me to use the Arduino print() method. However, since print() depends on the Arduino libraries. you probably cannot use it. (Note: That class also provides some other methods to display numbers in different formats).

For that reason you should probably focus on these methods that are implemented in the [TM16xx base class](https://github.com/maxint-rd/TM16xx/tree/master#basic-usage):
 - clearDisplay();
 - setDisplayToDecNumber(int nNumber, byte bDots=0);

Having those functions is probably sufficient to implement your basic needs. As stated before you should look into the [TM1640Anode class](https://github.com/maxint-rd/TM16xx/blob/master/src/TM1640Anode.h) to see how I implemented common anode support.

You can look at this [TM1640 example](https://github.com/maxint-rd/TM16xx/blob/master/examples/TM1640_basic/TM1640_basic.ino) which uses the setDisplayToDecNumber function.

beginner140BB, 2023-05-12:
Currently i am able to display single seven segment display.
what should be done to display multiple seven segments?
common anode 
[image]
when i have gone through the basic api in library i have not not seen how to control multiple displays at a time.

any api or any suggestion from your side?

maxint-rd, 2023-05-12:
Somehow your reply went missing. You wrote this:

> Currently i am able to display single seven segment display. what should be done to display multiple seven segments?
common anode 
[image]
when i have gone through the basic api in library i have not not seen how to control multiple displays at a time.

It seems you didn't look into the implementation of TM1640Anode, although I pointed to it three times before!
Let me try one more time to give you some further explanation.

To control multiple digits in common anode setup, you need to write the required data to multiple addresses. Look at table 9 on page 7 of the [TM1640 datasheet](https://github.com/maxint-rd/TM16xx/blob/master/documents/LED%20driver%20TM1640%202014-11-20-15-36-028.pdf). That table gives a nice overview of the display memory in relation to the GRD and SEG pins.
For common cathode, all segments of a single display would be in the same address of the display memory. In common anode setup this is different. All segments of a display connected to SEG1 are connected to a different GRD, meaning the same bit, but at different addresses, For SEG1 and GRD9-GRD16 it is bit B0 of addresses 08H-0FH. To set the segments of the display connected to SEG2, you need to modify bit B1 of the same addresses.

To easily manipulate multiple  common anode digiits in the implementation of TM1640Anode. I used an offscreen bitmap (a byte array) to be used as manipulatible memory. Read that code to see how... 
In [TM1640Anode.h](https://github.com/maxint-rd/TM16xx/blob/master/src/TM1640Anode.h) you see this definition:
```uint16_t bitmap[TM1640Anode_MAX_POS];       // store a bitmap for all 8 digits to allow common anode manipulation```

In [TM1640Anode.cpp](https://github.com/maxint-rd/TM16xx/blob/master/src/TM1640Anode.cpp) you can find this code in the function TM1640Anode::setSegments16(uint16_t segments, byte position)
```//update our memory bitmap
this->bitmap[position]=segments;

// Transpose the segments/positions to counter Common Anode connections and send the whole bitmap 
for (byte nSeg = 0; nSeg < _maxSegments; nSeg++)
{
 	byte nVal=0;
   	for(byte nPos=0; nPos < TM1640Anode_MAX_POS; nPos++)
	{
		// Assume 1st digit to be connected to SEG1, 2nd digit connected to SEG2 etc
		nVal |= (((this->bitmap[nPos] >> nSeg) & 1) << (nPos));
	}
	sendData(nSeg, nVal);
}
```

As you can see, every time a single digit is changed, all addresses are sent to the chip. I hope this makes it more clear. Unfortunately I have no means to take part in your project. I'm assuming certain level of programming skills, so if this explanation isn't sufficient, please hire a skilled programmer. Good luck!

beginner140BB, 2023-05-13:
@maxint-rd I just wanted to take a moment to express my sincere gratitude for the help and your valuable time.

Now i am able to control different segments and grids accordingly,as you stated the problem here i faced is not able to interpret the relationship table.

[image]

that B0-B7 values are used to control as 0x01(00000001) will turn on segment 1 and 0x02(00000010) will turn segment 2 0x03(00000011) will turn on segment 1 and 2 i think now i am right.