maxint-rd/TM16xx #38: Problem while using this library with ESP32

thetechguy2016, 2023-07-05:
Hi 

i am using this Library with ESP32 with GPIO-23 as DIN and GPIO-19 as SCLK
the segment is Common Anode.
i tried using the TM1640Anod.h also but its not working.

`
#include <TM1640Anode.h>

#if !defined(LED_BUILTIN)
#define LED_BUILTIN 2
#endif

// Define a 4-digit display module. Pin suggestions:
// ESP8266 (Wemos D1): data pin 5 (D1), clock pin 4 (D2)
// ATtiny44A: data pin 9, clock pin 10 (LED_BUILTIN: 8 on ATTinyCore)
TM1640Anode module(23, 19,12);    // data, clock, 4 digits

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
  //module.setDisplayToString("11111111");
  module.sendAsciiChar(1,'b',true);
  delay(500);                        // wait
  //module.clearDisplay();
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  //module.setupDisplay(true, 7);   // sometimes noise on the line may change the intensity level
  int nTime = ((millis() / 1000) / 60) * 100 + (millis() / 1000) % 60; // minutes+seconds as integer
  //module.setDisplayToDecNumber(nTime, _BV(4)); // display dot on digit 4
  module.sendAsciiChar(2,'2',true);
  delay(500);                        // wait
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  //module.setDisplayToDecNumber(nTime, _BV(3)); // display colon on digit 3
  module.sendAsciiChar(3,'1',true);
  delay(500);                        // wait
}
`

Did i miss something please let me know.

Thanks

maxint-rd, 2023-07-06:
Hello again,

Today I tested a similar configuration in this setup:
 - Win11, Arduino 2.1.0 using a UM TinyPICO ESP32 board, 
 - fresh installation of the TM16xx library via the library manager, using the released version (0.5.2110)
 - TM1640_basic example, modified to your sketch, using your pinout (DIN=GPIO23 and CLK=GPIO19), 

At first try I used the unmodified example with your pinout. Segments did light up and followed the instructions in the sketch. In Setup() I added a call to setSegmentMap() to match the segment-wiring of my display.
Then I made the modifications to use the sendAsciiChar call as in your code. The characters showed up as coded and I then used characters 'x' and '$', to test the alpha numeric segments of the display. They all worked fine.

Subsequently I tested the same board with a different IDE:
- Win7, Arduino IDE 1.8.16, ESP32 board TinyPICO
For that test I used a modified version of the example sketch [TM16xxDisplay_ScrolledPrint](https://github.com/maxint-rd/TM16xx/tree/master/examples/TM16xxDisplay_ScrolledPrint), using the TM1640 on the same pinout in combination with two other chips: a TM1668 and a HT16K33, both fitted with a 15-segment alpha numeric display, showing text scrolling over the three displays. Although that code is still under development for a future version of the TM16xx library, this combination also worked fine on that particular ESP32 board. 

In conclusion I must say I am currently unable to reproduce your issue. Can you provide more information and give reply to the questions and suggestions posed in my previous message?

maxint-rd, 2023-11-23:
Closed due to no response...

closed 2023-11-23 by maxint-rd