maxint-rd/TM16xx #39: Hello Any help on TM1621

Server4hassio, 2023-08-27:
How to control TM1621 on arduino. i am noob arduino and i have to control Sonoff THR316 Display by arduino. Its working tasmota and esphome. Help me pls

maxint-rd, 2023-08-27:
Hello, unfortunately I'm afraid I have to disappoint you a bit. The TM1621 is not a LED driver, but an LCD driver. My TM16xx library only supports LED drivers. I don't have any TM1621 chips in my collection (yet) nor any modules that feature a TM1621. For that reason I won't be able to help you directly.

However, I did some quick googling and perhaps these findings are helpful:

On the esphome website I did find [some information](https://esphome.io/components/display/tm1621.html) on the TM1621 support for your THR316 module and a link to the [source code](https://github.com/esphome/esphome)

More interesting might be [this informative video](https://www.youtube.com/watch?v=vDbWtbj-Lj4) and [this Arduino code](https://github.com/emsyscode/TM1621). It shows another display using the same TM1621 chip and the maker provides his Arduino code used to drive all segments of his display, I think that by studying that code, and reusing it with perhaps small modifications, you may be able to control your module with Arduino.

Nevertheless, thank your for bringing my attention to this chip. If I ever encounter such a chip or a module featuring the TM1621, I can add it to my collection and perhaps one day add support for it to my library. I will then reopen this issue to let you know. For now I wish you best of luck with your project.

maxint-rd, 2024-11-28:
Hello @Server4hassio, I like to mention that some time ago I finally purchased some TM1621D chips and hope to add support for the TM1621/TM1621D to the TM16xx library. In issue #59 you can follow my recent endeavors into this and I will first post any progress I make with my experiments there. 
Once support for TM1621 is complete I will also update this issue. (Reopened for now).

closed 2025-05-07 by maxint-rd