maxint-rd/TM16xx #41: Another TM1652 device

kymodoke, 2023-10-12:
Hello, this is not an issue, but a message to report the usage of TM16xx library on a device.

The **Xiaomi XIAO AI Smart Alarm Clock** (product code: zimi.clock.myk01) features a nice 7-segments display with a TM1652 chip that drives it.

History:  
_I bought that alarm clock for 3€ on a flea market with the intention to hack/modify/replace the embedded linux sytem on its mainboard (ARM board). It has many fetaures I don't like: basically it's a black-box puppet computer controlled from mainland China's cloud with wifi, bluetooth gateway and microphone, AI assistant (like Google, Alexa) but that speaks chinese only, and it need a smartphone app to do anything else... I gave up the project to hack the mainboard, because no time nor brain slot available...._  

But the 7-segment display is really nice and I wanted to repurpose it. So I digged further this strange 1 pin controlled and found out the controller is TM1652, then a web search led me to this repository.

[image]

Connections on the display are VDD - SDA - GND, connected on the mainboard to VCC - EN - GND. I measured working voltage to 4.6V.

So I reconnected the display to a tiny ESP32 board (Seeed Studio XIAO ESP32C3), and tested the display with "TM16xx_setSegments.ino" example. It does work, all segments are lighting up, except the middle colon.

[image]

So now it's a repurposed as a simple NTP clock, without all the nasty stuff, and with a K.I.S.S. approach.
It connects to wifi just at start up and then once a day to sync time to NTP, then switch the wifi off. Low luminosity between 21h and 6h. Perfect for my mother bedroom.

[image]

Thanks for TM16xx and the addition of TM1652

maxint-rd, 2023-10-13:
Excellent!
Thank you for sharing this nicely illustrated experience. It's interesting to see the various devices in the wild that use a TM16xx chip. I get inspired by reading how people find room for improvement or reuse. Congratulations on  your result!

If you find it interesting; amongst the other issues you can find some stories of like-minded people and their gadgets, such as the TM1680 based bluetooth speaker.

(For me K.I.S.S. is always the challenge. Often I overwhelm myself with a flow of would-be-nice yet not really essential features that keep me from completing things)

maxint-rd, 2024-06-26:
Closed as no further action is expected. I've added your device to the [list of real-world TM16xx devices](https://github.com/maxint-rd/TM16xx#real-world-devices-using-a-tm16xx-chip). If you ever encounter another TM16xx deviice, I'd appreciate any info you can post. Thanks again for your contribution!

closed 2024-06-26 by maxint-rd