maxint-rd/TM16xx #42: Halt when useing RP2040 with TM1652

YeWuChen, 2023-10-31:
I tried using RP2040 to drive TM1652, but the example would crash and Windows would prompt an unrecognized USB devices.
But I can use hardware serial to driven TM1652,  just use 'Serial2.begin(19200,SERIAL_8O2);' and set the TX pin to hardware serial pin, just 
pay attention to the initialization, need to add a about 6ms of delay between the cleardisplay and send command.

Arduino IDE version: 1.8.19
RP2040 version: 3.6.0 by Earle F. Philhower, III

And there's my code:
[my_TM1652.zip](https://github.com/maxint-rd/TM16xx/files/13211219/my_TM1652.zip)

It would be great if this library could be used hardware serial with all the good features of TM16XX librariey！

maxint-rd, 2023-10-31:
Hello @YeWuChen , 
Thank you for submitting this issue. It's great to read you found a workaround for the issue you encountered.

Some time ago I encountered [stability issues with the TM1640 on RP2040](https://github.com/maxint-rd/TM16xx/issues/34#issuecomment-1476606214). (See the short video clips in those posts). Unfortunately even after using a better scope, I wasn't able to determine the cause. I'm definitely no expert on the RP2040. Perhaps your findings can help to implement a solution of some sort. From what I recall I've not done any testing using hardware serial on the RP2040. Also I have not tested all TM16xx chips in my collection with the RP2040.

I think I need to do some more testing on the RP2040 to see what issues I can reproduce and if possible fix. Unfortunately my work on the TM16xx library is limited to the spare time I have available. Because of that it may take quite some time before I can respond with my findings. I will keep this issue open until I've done so.

YeWuChen, 2023-11-01:
Hello @maxint-rd ,
Thank you very much for your reply! I used the two examples TM16xx setSegments and TM16xxDisplay_Print, and modified the header files and classes in the examples to TM1652, and set the pin and parameters.Both examples resulted in the RP2040 halted，So I didn't try other examples, but these two examples with modified header files and classes to TM1652 programmed into my Mega2560 worked fine！

With best wishes.

maxint-rd, 2024-06-26:
FIY: During recent work on developing a generic class to support untested TM16xx chips, I used the RP2040 to test things. I fixed the halting issue for RP2040 (and also some other MCUs) by introducing a begin() method with implicit initialization that replaces blocking time related calls in the constructor. You can find that fix in [this repository branch](https://github.com/maxint-rd/TM16xx/tree/add-generic-class-to-support-(almost)-all-TM16xx-chips). It is expected that in due time that fix will be merged into the main branch.

maxint-rd, 2024-12-23:
FYI: the TM1652 has been updated and code from the other branch was incorporated. Still not using hardware serial though and haven't tested with RP2040 yet.
I'm not sure yet what approach I would take to have a flexible implementation that supports both hardware and software serial. Most processors have limited availability of hardware and then it's used for debugging/programming  At the moment this isn't at the top of my priority list, but I'll keep this issue open for now.

YeWuChen, 2024-12-25:
Thank you very much for your efforts! However, my board is not with me right now, so I have no way to test the updated library on the RP2040 for the time being, I will test it once I get a chance. Thank you very much again!