maxint-rd/TM16xx #47: How do you control the TM1638 discrete LEDS?

zapdogster, 2024-02-21:
More TM1638 libraries I've seen have functions to drive the row of discrete LEDs that are along the top of the board. I don't see an equivalent in this library. Am I missing something?

maxint-rd, 2024-02-21:
Hello @zapdogster , it kind of depends on the module that you're using. Can I assume your module is labeled LED&KEY has 8 buttons below the digits and 8 discrete red LEDs above the digits?

Note that there are a few different display modules, all featuring the TM1638 chip. Some modules look similar but may have different wiring. My first module is the LED&KEY module mentioned above. I believe there is also a similar LED&KEY module having 8 discrete bi-color red/green LEDs above the digits. I also have a module labeled QYF-TM1638 that has 4x4 buttons, without discrete LED's. The is also a module with different wiring, using common anode digits instead of common cathode.

This library has its origin in the  library by rjbatista, which explains why some basic examples and documentation may be lacking. For the discrete LEDs the methods setLED() and setLEDs() are implemented:
```
    /** Set the LED at pos to color (TM1638_COLOR_RED, TM1638_COLOR_GREEN or both) */
    virtual void setLED(byte color, byte pos);
    /** Set the LEDs. MSB byte for the green LEDs, LSB for the red LEDs */
    virtual void setLEDs(word led);
```
If you want to drive individual digit LEDs you can also try the method setSegments16().

zapdogster, 2024-02-21:
Thanks for the reply. Yes the module I have is the one you describe as "LED&KEY". I will give those methods you have listed a try. Thanks!

maxint-rd, 2025-04-10:
Assuming all is okay I will close this issue now. Feel free to post further questions/issues if needed.