maxint-rd/TM16xx #48: Button numbering question for TM1638 LED&KEY module (8 LEDS, 8 digits, 8 Buttons)?

zapdogster, 2024-02-21:
I am testing this library by running the TM16xxButtons_clicks.ino example. I was expecting the Button numbers to be reported as either 0 thru 7, or 1 thru 8. However this is what I see:
ButtonPressed    DisplayedValue   fnClick Reported Button
1                             10000              Button 16 Click
2                             40000              Button 18 Click
3                           100000              Button 20 Click
4                           400000              Button 22 Click
5                             20000              Button 17 Click
6                             80000              Button 19 Click
7                           200000              Button 21 Click
8                           800000              Button 23 Click

I think I see where these values are created: TM1628::getButtons(void) in TM1628.cpp

Is there something simple I'm missing to get the values I was expecting? Or do i have to do a manual translation to get the expected values?
Thx!

zapdogster, 2024-02-21:
Not sure this is the best approach, but I just did a brute force translation. Pls let me know if there is a better way:

// **********************************************************************
// Function to translate button hex code returned to expected 0-7 value
int fnTranslateButtonCode (uint32_t dwButton)
{
  int i_ButtonTranslated  = dwButton >> 16 ;

  switch (i_ButtonTranslated) {
    case 0x01:
      i_ButtonTranslated = 0 ;
      break ;
    case 0x04:
      i_ButtonTranslated = 1 ;
      break ;
    case 0x10:
      i_ButtonTranslated = 2 ;
      break ;
    case 0x40:
      i_ButtonTranslated = 3 ;
      break ;
    case 0x02:
      i_ButtonTranslated = 4 ;
      break ;
    case 0x08:
      i_ButtonTranslated = 5 ;
      break ;
    case 0x20:
      i_ButtonTranslated = 6 ;
      break ;
    case 0x80:
      i_ButtonTranslated = 7 ;
      break ;
  }
  return i_ButtonTranslated ;
}

maxint-rd, 2024-06-26:
Closed as no further action is expected. Hopefully my last response was helpful. Feel free to reopen if you have further (related) questions.