maxint-rd/TM16xx #51: Request: Brightness scaled [0-255] instead of [0-7]

ArnieO, 2024-05-11:
One byte is commonly used for setting the brightness of LEDs, with value ranging from 0 to 255. 
This library sets brightness in seven steps, from 0 to 7.

In my project I use an ambient light sensor to set the intensity of both 7-segment displays and LED strips on the same enclosure surface. It would be very useful to be able to adjust the 7-segment brightness in finer steps from 0 to 255.

Maybe the `setIntensity()` function could get an additional parameter to accommodate this request?

maxint-rd, 2024-05-11:
Hello @ArnieO , I have to disappoint you, Unfortunately almost all TM16xx chips are limited to 8 levels of intensity/brightness. As you can see on page 7 of the TM1618 datasheet, that chip also only has 8 levels.

There are only a few exceptions that I know of:
- The TM1652 allows for setting drive current so for that chip setIntensity has an additional parameter, but for consistency the 16 brightness levels of that chip are converted to 8 levels. 
- Another interesting exception is the TM1626A, which offers three separate LED outputs with PWM. I guess nice to drive an RGB LED, but I think it's totally unrelated to the 7-segment display driver.
- There are some other weird TM16xx family members, such as the TM1648A, which supports driving a buzzer; and the TM1643 which has a totally different Charlieplexing way of connecting LED's.

For this library I think I'll stick to the most common things. The universal chip supporting class that I've been working on supports all common features for those chips that use  2 or 3 wire interfaces, allowing for dare devils to connect chips that I don't have in my collection (yet). I hope to publish that class later this month or the next.

ArnieO, 2024-05-11:
OK - so it is a limitation of the TM16xx (in my case TM1618) chip, understood. 

Thank you for your rapid clarification!

closed 2024-05-11 by ArnieO