maxint-rd/TM16xx #52: AIP1629B/TM1629B support

Voha888, 2024-05-27:
[TM1629A_V2.0.pdf](https://github.com/maxint-rd/TM16xx/files/15456398/TM1629A_V2.0.pdf)
I want to connect this chip model. And as far as I understand, I just need to change the existing header file to accommodate the number of signal lines 1629B?

1629B controls a display like this, I want to write a test sketch to determine which segments, which data lines correspond to
[image]

maxint-rd, 2024-05-27:
Hello, thank you for your post and for including a picture of the display module.

Unfortunately I don't have any to the different TM1629 chips in my collection. For that reason I have not made a chip specific class for the TM1629B. 

Although that chip is similar to the TM1638, the TM1628 and the TM1668, there are differences that most likely will result in limited functionality. For you it may be good to just give these a try. Using an incorrect chip class will not break the hardware.

In the past weeks I have been working on a generic class to allow adventurous developers to experiment with chips that I have not tested. It contains definitions for all TM16xx chips and generic code supporting different type of connections, various display memory layouts and key scanning methods.

That class is still under development, but if you are willing to spend I could publish it so you can give that class a try. I think it could provide a good matching support for the TM1629B.

To illustrate; this is an example of the chip specific definitions used in that generic class:
```
#define IC_TM1628A if_ctrl_tm16xx {TM16XX_IFTYPE_3WSER, 5, 7, TM16XX_IC_SEGMUX_3 | TM16XX_IC_SEGBYTES_2, 0x3BFF, 5, 0x1B}  // same as TM1628, but with internal pull-up/down resistors
#define IC_TM1629  if_ctrl_tm16xx {TM16XX_IFTYPE_4WSER, 5, 8, TM16XX_IC_SEGMUX_0 | TM16XX_IC_SEGBYTES_2, 0xFFFF, 4, 0xFF}  // SxG: 16x8 CC/CA, K:8x4
#define IC_TM1629A if_ctrl_tm16xx {TM16XX_IFTYPE_3WSER, 5, 8, TM16XX_IC_SEGMUX_0 | TM16XX_IC_SEGBYTES_2, 0xFFFF, 0, 0x00}  // SxG: 16x8 CC/CA
#define IC_TM1629B if_ctrl_tm16xx {TM16XX_IFTYPE_3WSER, 5, 8, TM16XX_IC_SEGMUX_0 | TM16XX_IC_SEGBYTES_2, 0x3FFF, 4, 0xBB}  // SxG: 14x8 CC/CA, K:8x2
#define IC_TM1629C if_ctrl_tm16xx {TM16XX_IFTYPE_3WSER, 5, 8, TM16XX_IC_SEGMUX_0 | TM16XX_IC_SEGBYTES_2, 0x7FFF, 4, 0x88}  // SxG: 15x8 CC/CA, K:8x1
#define IC_TM1629D if_ctrl_tm16xx {TM16XX_IFTYPE_3WSER, 5, 8, TM16XX_IC_SEGMUX_0 | TM16XX_IC_SEGBYTES_2, 0x0FFF, 4, 0xFF}  // SxG: 12x8 CC/CA, K:8x4
```

Voha888, 2024-05-27:
I found a person on the Internet who used the TM1628 and TM1668 libraries. He simply changed the number of segments from 14 to 16 and it worked.
The display in the photo is with a friend of mine who lives far away. When he receives his Arduino board, I will be able to test this in practice, and will try to add the corresponding header file to your repository

[image]
[image]
[image]
[image]

Voha888, 2024-06-09:
Good day! I'm very sorry, I didn't read your first message carefully. I thought I could find your generic class in the library, but I didn’t find it. Could you publish it? I had the opportunity to test it on a real display. Thank you

maxint-rd, 2024-06-09:
> That class is still under development, but if you are willing to spend I could publish it so you can give that class a try. I think it could provide a good matching support for the TM1629B.

> Good day! I'm very sorry, I didn't read your first message carefully. I thought I
could find your generic class in the library, but I didn’t find it. Could you
publish it? I had the opportunity to test it on a real display. Thank you

Just saw your message. I was indeed waiting for a reply and while doing so did some more testing of the generic class on the RP2040. Thank you for your response. I will upload the required changes asap, and add a comment here when done.
If not later this evening, I will do so in the next few days. To be continued...

Voha888, 2024-06-09:
Hello!  I conducted some experiments and determined which TM1629b registers include which segments.  I didn't understand how your library works, so I used my own code.  So I deleted my post - I thought I'd better present the finished result.  It works, but has not been finalized yet.  I will use Arduino Nano for control.  But I have RP2040.  and I can conduct any experiments you suggest.  

Here is my draft for this display with AiP1629B.  Where P0-P7 are the byte numbers, and Б0-Б7 are the bit numbers in each byte
[image]

maxint-rd, 2024-06-09:
Hello @Voha888 , thanks for your elaboration. Doing something yourself is quite educational, so I recommend that at all times. I still will upload my changes and if you don't mind, perhaps you can give it a go. I also like to know what you didn't understand. Was it my code? Or didn't you know how to use it? 

This library aims to support all TM16xx chips. Such goal and the long history of the library may have introduced some complexities you won't have when developing for a specific chip from scratch. The generic class that I will upload in a few moments tries to address all commonalities and may even have more complexities as a result. I always try to write readable comments, with sufficient comments, but other documentation may be lacking a bit. Ah well... perhaps it will get better sometime...
(for now I'll first focus on uploading the changes,...)

maxint-rd, 2024-06-09:
Hello @Voha888 , I added a branch for testing the generic class TM16xxIC. You can find it [here](https://github.com/maxint-rd/TM16xx/tree/add-generic-class-to-support-(almost)-all-TM16xx-chips).

This is the constructor code I used to test this class with a TM1640 on the RP2040:
```
#include <TM16xxIC.h>
TM16xxIC module(IC_TM1640, 18, 19);    // For ESP8266/WeMos D1-mini: DIN=D7/13/MOSI, CLK=D5/14/SCK; RP2040 PICO: DIN=18, SCK=19
```

Please note that although for now I tested it only on the RP2040, it should also work on other boards such as Arduino Nano. As you probably can guess I don't have the AiP1629B in my collection either, so I cannot tell how compatible it is with the TM16xx series.  

In the include file TM16xxIC.h you can find a list of all TM16xx chips that I know of. I studied their datasheets to determine their specifics in terms of SEG/GRD layout and display/key memory. I included definitions for all those files and made comments about the chips that I have tested this generic class with. Most 2 and 3 wire chips should be supported. Some chips (eg. TM1650, TM1652) are too specific to include in the generic class. For those chip specific classes are available or may be added in the future,

I will do more testing on many other MCU's, such as ATtiny85, ATMEGA328P, LGT8F328P, CH32V003, ESP8266, ESP32, but that may take some time. 
The library is probably too large for the smallest MCU's (such as ATtiny13A) but should work on most others that have an Arduino core available.

maxint-rd, 2025-01-12:
FYI: the generic class has been added into the main branch.

Voha888, 2025-01-12:
Thank you! I tested it and created an example back when we were talking. But I forgot to publish it. I will add my example to the examples as soon as possible