maxint-rd/TM16xx #54: USE TM1624 7x11, what should I do?

rx-ted, 2024-09-29:
USE TM1624 7x11, what should I do?

maxint-rd, 2024-09-29:
Hello, the latest release of the TM16xx library has no explicit support for the TM1624 (yet).  In issue #24 you can find another fellow with a TM1624, but at that time I couldn't offer much code to help. Now maybe I can...  :-)

But first, can you tell me a bit more about your project? Are you using a bare chip or is it part of a existing module? What kind of LED display is attached? Is it Common Cathode or Common Anode? What MCU will you use?

Some time ago I was able to add the TM1624 to my collection, Since then I have worked on a generic class that aims to support *almost all* TM16xx LED driver chips. That generic class intends to allow adventurous developers to experiment with chips that I may not have tested myself. It contains definitions for all TM16xx chips and generic code supporting different type of connections, various display memory layouts and key scanning methods. 

At the moment that added class has not been thoroughly tested, but I've tested it to work for my TM1624. 
It may help you too in your TM1624 project; assuming you're willing and able to experiment and spend time on using undocumented code.

I have published my work in progress in [this repository branch](https://github.com/maxint-rd/TM16xx/tree/add-generic-class-to-support-(almost)-all-TM16xx-chips/src). The class is named TM16xxIC and it is used in similar fashion as the regular TM16xx classes. The generic [TM16xxIC class](https://github.com/maxint-rd/TM16xx/blob/add-generic-class-to-support-(almost)-all-TM16xx-chips/src/TM16xxIC.h) uses a chip specific define in the constructor to specify the chip characteristics.  In [this post](https://github.com/maxint-rd/TM16xx/issues/52#issuecomment-2156763990) you can see how I used it for another chip.

In your sketch you need to include [TM16xxIC.h](https://github.com/maxint-rd/TM16xx/blob/add-generic-class-to-support-(almost)-all-TM16xx-chips/src/TM16xxIC.h), which has the chip definition ```IC_TM1624```. 
The TM1624 supports 14x4 - 11x7 segments/grids, no keys. 

I don't have a ready made example for using this class yet, nor proper documentation. The main examples I use for testing are [TM16xxDisplay_Print](https://github.com/maxint-rd/TM16xx/tree/master/examples/TM16xxDisplay_Print) and [TM16xx_setSegments](https://github.com/maxint-rd/TM16xx/tree/master/examples/TM16xx_setSegments) / [TM16xx_setSegmentsDebug](https://github.com/maxint-rd/TM16xx/tree/master/examples/TM16xx_setSegmentsDebug). If I remember well, I believeI tested my TM1624 chip with a 4-digit 7-segment CA display.

BTW. To support common anode SEG/GRD configuration that branch also has a [TM16xxICanode](https://github.com/maxint-rd/TM16xx/blob/add-generic-class-to-support-(almost)-all-TM16xx-chips/src/TM16xxICanode.h) class. Those generic classes have a slightly larger footprint, but still may be useful to support the various other chips in the TM16xx family.

rx-ted, 2024-09-29:
Thank you for the information you provided earlier. I will be testing with the Arduino UNO board after the National Day holiday.

My current hardware setup:
- STM32F103C8T6 (Blue Pill board)
- TM1624 chip
- 11x7 LED matrix (11 segments are positive (+) and 7 bits are negative (-))

**Issue:**
- The circuit is functioning, and the LEDs are displaying something.
- However, the output is not as expected, and I suspect there may be an issue with the timing sequence.

Thanks again for your help! I will share the results after testing.

maxint-rd, 2024-09-29:
Hello, unfortunately I have no test setup for the STM32F103C8T6 myself, but if I recall well other people did test that environment successfully with the library. I'm looking forward to read about your findings. 
I have tested several TM16xx chips with other processors that run at higher clock speeds than the Arduino Uno (eg. CH32V003, ESP8266 and RP2040) . Most chips worked fine on 3v3 when using red LEDs, some required 5V or a capacitor. Fast speeds often required better breadboard connections and short wires.

If the LEDs are displaying something, I suggest you to try the [TM16xx_setSegments](https://github.com/maxint-rd/TM16xx/tree/master/examples/TM16xx_setSegments) example first to check if the segment order is according expectations. Replace the ```setSegments()``` call by ```setSegments16()``` to test more than 8 segments. In a normal setup segments A-G would be connected to SEG1-7. If you use a deviating segment order, the TM16xxIC class can use a segment map provided using ```setAlphaNumeric()```.
Good luck!

rx-ted, 2024-10-13:
Thanks. I wrote a simple port of [tm1624](https://github.com/rx-ted/TM1624), and it works normally on the stm32f103c8. The underlying principle is based on the library you provided, so there's actually no difference.

closed 2024-10-13 by rx-ted

maxint-rd, 2024-10-13:
Excellent. Good you got it to work for you. Perhaps in the future I'll do some blue pill testing. At the moment I'm into the CH32 family that has many commonalities.

maxint-rd, 2025-01-02:
Update: TM1624.h was added to offer support for the TM1624 via the TM16xxIC class. I have tested the TM1624 myself on the CH32V003 in 14x4 mode with an 15-segment alphanumeric LED display.

The TM16xxIC class was merged in the main branch, and the experimental branch will be removed at a later moment.

rx-ted, 2025-01-02:
Cool! Thanks for your updating.