maxint-rd/TM16xx #56: TM1637 - Buttons don't work on esp32

ANLCRAW, 2024-10-08:
Thanks for your library! I’ve been using it with the ATmega328P and costom pcb, but now I'm switching to the ESP32 and after 24 hours of struggling, I discovered that I need to change 'delayMicroseconds(5)' to 'delayMicroseconds(330)' in tm1637.cpp to make the buttons work. Else i get de default display printing of 0x8000 / no buttons / 1111-1111. I'm using de tm16xxButtons_clicks example.  Anyone else have experience with this? I'm using a MH-ET LIVE

maxint-rd, 2024-10-08:
Hello @ANLCRAW - thanks for taking the effort to find the root cause of your issue and for posting this.
You're the first I know to encounter this specific case. I have tested the library on ESP32 (on my TinyPico board), but probably not all with combinations of displays. In some cases I did need to change introduce delays to facilitate faster processors. I think I need to setup a test environment to test this myself and see if I can reproduce your findings.

The MH-ET LIVE board looks a lot like the D1 mini. I do have the ESP8266 variant of that board, but I don't think I have it with an ESP32. However, I think my TinyPico should suffice for testing this. 

Can you tell me what version of the Arduino IDE and which version of the ESP32 core you are using? At what clock speed is your ESP32 running? In some tests I ran my ESP8266 at 80MHz and the RP2040 at 133MHz. I can't remember what speed I ran the ESP32 when testing. I'm not sure if 240MHz is the default speed/.

I don't know when I can find some time to test this, but I will keep this issue open and post my findings here.

ANLCRAW, 2024-10-08:
Super thanks for your quick reply. The "MH-ET Live Minikit" resembles a D1 Mini, but I have the "MH-ET Live devKit," which is more like a NodeMCU, though that probably doesn't make much difference. The ESP is an "eps-wroom-32," and to my knowledge, it's running at 240MHz. I am using the "espressif esp32" core version 3.0.5 (MH-ET LIVE ESP32DevKIT) with the latest Arduino IDE version 2.3.3. ESP32 runs at 80MHz.

As I’m typing this, I am removing the espressif esp32 core (https://github.com/espressif/arduino-esp32) and trying the core provided by MHEtLive (https://github.com/MHEtLive/arduino-esp32), hoping that it might make a big difference. I will let you know my findings.

The wires between the ESP and the display are no longer than 2 centimeters. The TM1637, buttons, and display are all on a PCB.

I hope to find a solution because going from 5ms to 330ms is very noticeable when navigating through the menu. I’ll keep searching. It’s likely not your fantastic library causing the issue but something on my end. I’ve already ordered new TM1637 chips to rule out the possibility that the chips are the problem.

maxint-rd, 2024-10-08:
Thank you for the swift response to my questions. The delay is used at multiple points, so an increase from 5us to 330us will add up. (I assumed you meant to write 330us instead of 330ms (milliseconds) as 330ms would definitely be unworkable). You mention both 240MHz and 80MHz. Was the issue the same when running at the slowest speed? 

One thing I didn't ask yet is about the color of the leds and the supplied voltage. Most TM16xx chips work fine on 3v3, but I did notice some issues when using blue led display instead of red. The blue leds require higher voltage and may require a level shifter.

When I can find the time I'll try to match your setup. Some years ago I made a small [TM1637 matrix PCB](https://github.com/maxint-rd/arduino-modules/wiki/07-TM1637-6x5-LED-matrix) that allows the addition of [three buttons](https://github.com/maxint-rd/arduino-modules/wiki/08-3x1-buttons-for-TM16xx). I used it to make [this clock](https://youtu.be/FfmRCfF5qQI), Before making those PCB's I modified a TM1637 4x8 segment display module into a [mini LED&KEY module](https://youtu.be/8PaEZGtskVs?si=5Se1I1gZHG5h6LkC).  I still have plenty TM1637 chips to do some experiments.

Your chip may be defective or perhaps just less reliable at higher speeds. On one occasion I had a TM1640 matrix module on the RP2040 which gave [random glitches](https://github.com/maxint-rd/TM16xx/issues/34), but only on the RP2040, not on slower processors. I tried everything including analysis with a scope, but couldn't find the issue. Sometime later I tried again, but then also tried another matrix module. Guess what: one module displayed the glitches while the other module worked just fine. In [that test](https://github.com/maxint-rd/TM16xx/issues/34#issuecomment-2125623415) it seemed to be a combination of a chip less reliable at higher speed plus iffy breadboard connections.

ANLCRAW, 2024-10-08:
Apologies for my little mistakes; they have been corrected. I did indeed mean microseconds (µs) and not milliseconds (ms). Although I wasn’t very clear about it, what I meant is that the ESP32 is a 240MHz chip, but I’m running it at 80MHz. Lowering it to 40MHz makes no difference. Using the MH-ET core also made no difference, so I’m back to using the Espressif core 3.0.5.

This is what my current PCB looks like: 
[image]

Good videos by the way... lekker op zijn hollands! 

I am almost one hundred percent sure that the issue lies with the TM1637 chips. The first two TM1637 chips I used came from a 7-segment DIY module, and I then bought a batch from AliExpress. The chips from the 7-segment module apparently don’t work on the ESP32, and the new batch doesn’t work on the Atmel328p. Viceversa.

While waiting for the new chips to arrive, for anyone who has had the same experience... not every chip is the same, something I was aware of but had never experienced so clearly. There are plenty of rejected chips being sold.

Thanks again for your response, and I’ll share my findings here in a few weeks.

maxint-rd, 2025-01-22:
Hello again; 
In the past months I've made some updates that should improve support on newer ESP cores, but I haven't tested them on the ESP yet (mainly tested with CH32) .
The changes include adding the generic TM16xxIC class to support many more TM16xx chips. I did some cleanup of the TM1637 class and added the alternative TM1637IC class, which uses that generic class. Note that due to implementation differences the button numbers will be different when using that class.

If you have time to test, could you please try out the ESP32 with both TM1637 and TM1637IC? I'm very interested to know if it makes any difference on your TM1637 chips.
I'm also interested to see if the new chips you ordered behave any better.

After some more testing on various platforms I intend to make a new release that has all changes. Thank you for your efforts so far.

maxint-rd, 2025-01-28:
Hello @ANLCRAW;

Congratulations! Good to read you found a solution. Interesting to see it works after you remove the recommended passives. 

Thank you very much for your kind help in testing the latest code and for reporting your findings. I still need to do more testing of my own, but it's great to read you've already tested your ESP32 and found no further issues. 

I understand your spare time is limited too, so I really appreciate your contribution. I wonder what the root cause is, but don't think I will be able to reproduce your issue. Just for the sake of completeness I will post a longer reaction below. There is no requirement to answer any of the questions I pose. Time is limited, so please only read on if you have nothing better to do... ;-)

=======  

Have you measured the voltage on the DIO/CLK  lines when idle? Perhaps your ESP32 board already has pull-ups/caps? 
Since you did get proper digits showing on the display, my first thought is that it could be something with the caps rather than the pull-ups.

I double checked the datasheet (v2.4) and saw the circuit on page 8 to indeed have the values you mention. Could it be that perhaps your capacitors were faulty or had the wrong value? Have you measured your capacitors after removing them from the PCB? Were they the correct value? (In some other experiments I noticed that adding a capacitor can be used to reduce noise, but may limit the speed of communication).

FYI: the popular 4-digit TM1637 Arduino module also has pull-up resistors and filtering caps (C1/C3 and 10K R1/R2), but that module has no buttons. Looking back at the video of my modified version of that board with added buttons, it appears I tested it with a D1 mini ESP8266 board. In that environment I encountered no such issues.

Is your C1 capacitor there for light measurement purposes? The datasheet circuit also shows some power support caps (104 and 100uF between VCC and GND). Your module doesn't seem to have those, but neither has the 4-digit Arduino module.

BTW. below the circuit diagram the datasheet has these notes:
> 1. filtering capacitor between VDD and GND should be arranged on PCB plate as close to TM1637 chip as possible to strengthen filtering effect. 
> 2. 100pF capacitor connected to the DIO, CLK communication port pull-up and pull-down can reduce interference to radio communications port. 
> 3.  Since blue-ray nixie tube break over step-down voltage is 3V, TM1637 power supply should be 5V

To conclude I just wanted to share this random remark about testing:
Some time ago I tested an earlier release of the library on Wokwi with the Arduino Uno simulator. Today I also added an [ESP32 example on Wokwi](https://wokwi.com/projects/421321782868516865) that uses the released library on a simulated TM1637 display. Once the release is updated I intend to retest to see if that still works. Although it only has the simulated TM1637; Wokwi has simulators for various ESP32 variants as well as other plaforms such as AVR, Pico RP204 and STM32. Having those there could be useful for testing different environments...

maxint-rd, 2025-02-01:
The new release seems to work. Tested on RP2040, CH32 and via simulator on ESP32 and AVR, 
I'll close this issue now, but feel free to add more info, reopen or post a new issue.

closed 2025-02-01 by maxint-rd