maxint-rd/TM16xx #57: release callack

alikanarya, 2024-10-14:
Hi, firstly I thank for your library.
I want to simulate tm1638 buttons as push puttons.
I coded and worked.
I attached release and click callback functions.
However, release fn is called __before__ click fn.
And this makes some awkward behaviour in my scenerio.
Could you please guide me how to call release callback fn after click fn?
Thanks

example monitor output when I pressed and released button 1:
relased button: 1
pressed button: 1

maxint-rd, 2024-10-14:
Hello, the behavior you describe is by design. Currently there is no callback for a button being pressed. Only after a button is released we can determine if it was a click, a double click or a long click.

The main goal for having a release callback is for fast actions when no click or double click needs to be detected while allowing checking if other buttons are still being pressed (think of a shift key).

The normal usage of what I think you intend to do is only have a click callback, which is indeed called right after the button was released. 

FYI: in case of a double-click the sequence is like this:
```
Button 2 release.
Button 2 release.
Button 2 doubleclick.
```

If you still want to act before the release is done (e.g. to give visual indication of a button being pressed), you could use the return value of `tick()`. The 32-bit return value indicates if buttons are pressed (value != 0) and which buttons are pressed (bit at button position is set). Is this explanation helpful?

maxint-rd, 2024-10-14:
I think in your mind you need to distinguish press from click. A click is press+release. A long click has long duration between press and release. A double click is two clicks within a short time.

Normally you call the tick() method numerous times at the top of your running loop(). The tick() method implements a state engine to determine the state of each possible button.  Only once released tick() can tell if a button was clicked once, twice or long. If you print millis() in the callbacks you can see the time between the various events.

Each time you call tick, it returns a 32-bit value containing the on/off value of 32 buttons maximum. Before returning it sets timers, updates the state engine and does callbacks. However when just pressing, tick() currently returns without doing a callback. Whether and when a specific button is pressed is stored in the state engine. Eventually tick() returns the current on/off value of all possible buttons. Does this clarify things?

alikanarya, 2024-10-14:
Thank for your support

maxint-rd, 2024-11-28:
You're welcome. I think I can close this issue now. If not, just let me know.

closed 2024-11-28 by maxint-rd