maxint-rd/TM16xx #59: Request - Add new chip to library - TM1621

JGG-Apps, 2024-11-24:
I'm interested to use a LCD display that run with TM1621 display driver, I was searching on the Internet and I found your project that already try to manage all TMxxxx chip in one library !

The LCD display I have is coming from an old portable radio. It have 8 characters contains 14 segments (see sample in attached files) + 14 individual pre-formatted text (like: Bass, Rock, Live, etc.). 

The board is using 7 pins to communicate, technically (VDD, VSS, VLCD and 4 COM), not 100% sure...

The TM1621 display driver can manage many configuration of LCD displays if I take a look to this video: https://www.youtube.com/watch?v=vDbWtbj-Lj4 where someone already start a independant project to re-use another LCD display with the same chip. He already put his source code on GitHub under another project, but he never completed. since 6 years. Link: https://github.com/emsyscode/TM1621

About the TM1621 display driver:

The Titan TM1621 Display Driver is a SSOP chip with 48 pins. He drive 32 segments outputs (SEG0 to SEG31) with 4 communication inputs (COM0-COM3). This chip is also available in LQFP format with 44 pins where 4 missing pins are internally managed, like 2 pins for  crystal (OSC0-OSC1), RD and IRQ are not available.

He work under nominal voltage of +5VDC (2.4V min-5.2V max)
The BZ pins are for a buzzer (piezo speaker)... These pins are not in use on my personnal board, no piezo.
The VLCD pin is for the brightness using a variable resistor between the VSS voltage (nominal +4VDC, using a variable 15Kohms to adjust)

The PDF Chinese guide for TM1621 is also available here: https://www.alldatasheet.com/datasheet-pdf/pdf/1131967/TITAN/TM1621.html

He probably use a similar communication signal with other TITAN drivers, but with more segments than almost other chips you already added in this library:

*** default minimal output segments ***
TM1616 (SEG1-SEG7) 7 segments
TM1618 (SEG1-SEG5) 5 segments
TM1620 (SEG1-SEG8) 8 segments
TM1621 (SEG0-SEG31) 32 segments *** my request ***
TM1628 (SEG1-SEG10) 10 segments
TM1630 (SEG2-SEG8) 7 segments
TM1637 (SEG1-SEG8) 8 segments
TM1638 (SEG1-SEG10) 10 segments
TM1640 (SEG1-SEG8) 8 segments
TM1650 (A-G) 7 segments
TM1652 (SG1-SG7) 7 segments
TM1668 (SEG1-SEG10) 10 segments
TM1680 (ROW0-ROW23) 24 segments 

On my LCD display, each of all 8 digit have 14 segments like that:
[image]

Sample of TM1621 chip soldered on the board:
[image]

Sample of the LCD display, 1 line with 8 characters (each character have 14 segments) + 14 individual tags... The technically 9 digits of 14 segments :
[image]

maxint-rd, 2024-11-24:
Hello @JGG-Apps, 
Thank you for submitting this issue with so many details and excellent information. I will probably have to read it once more to capture all details, but I don't want to keep you waiting too long, so here is already a brief answer. (Edit: sorry, it became larger than I planned...)

Some time ago someone else posted issue #39, with a similar question. At the time I didn't have any TM1621 chips, nor any modules featuring that chip. Although I still haven't found a LCD module, I recently did purchase some TM1621d chips. From what I saw the TM1621d is similar to the TM1621, but in a smaller SOP24 package (and perhaps less SEG/GRD pins). My plan is to try to connect one with a randomly found LCD panel, to see if I can manipulate it. If those experiments are successful I intend to add the TM1621 to the library.

I already provided support for various chips with more segments and for instance used a TM1640 in common anode mode and some others chips (eg. 14-seg TM1624) to drive a 16-segment alpha-numeric LED display. Support for 32 segments probably requires some  API changes, but I also want to add the TM1680 that as you stated can do up to 24 segments x 16 digits.

Indeed I aim to support all TM16xx chips eventually. In [this branch](https://github.com/maxint-rd/TM16xx/tree/add-generic-class-to-support-(almost)-all-TM16xx-chips) I added the generic class TM16xxIC to support (almost) all TM16xx chips. Currently it still is experimental code for those who dare. It includes untested support for chips not in my collection. See the list in [TM16xxIC.h](https://github.com/maxint-rd/TM16xx/blob/add-generic-class-to-support-(almost)-all-TM16xx-chips/src/TM16xxIC.h) where I added chip specifics as comments after each definition.

I already have added the TM1680 to my collection and saw it needs a bit more work to support it as it uses an I2C connection. I have something working for bare chips, but not quite ready for publication. I also did some special [experiments with a TM1680 bluetooth speaker](https://github.com/maxint-rd/TM16xx/issues/2#issuecomment-1411017036).

A first question about your module: you say you rescued it from an old radio. Did you already try that other code?
I certainly don't have that radio. Did you perhaps encounter some other display modules that have this chip? (If I can buy such module I don't have to find a LCD panel and figure out its connections).

maxint-rd, 2024-11-24:
Something more in the meanwhile: in [this YT video](https://github.com/MakersFunDuck/Common-Kitchen-Timer-With-LCD-Replacement-Board) you can see someone managed to connect a LCD Kitchen Timers (Egg timers) to the TM1621d. In the comments you can find some ideas we exchanged about a library implementation.  That person also published [his work on GitHub](https://github.com/MakersFunDuck/Common-Kitchen-Timer-With-LCD-Replacement-Board). 
If you have time perhaps you can see what it does for you. I'm not sure when I have enough spare time to dig a bit deeper...

JGG-Apps, 2024-11-25:
The old radio is dead anf I pulled out the LCD many years ago. I d'ont remember exact radio model... Anyway. I decided to figure out pin usage on the board connector... Then I removed the plastic case and I found very interesting information !!! You can see the picture.

4 pins are for voltage operation, where 2 pins are not connected to the TM1621 in my case... The manufacturer decided to power the backlight LCD directly. 2 others pins are for TM1621 voltage. Finally, 3 last pins are for data (CS, Clock, Data) !!!

This reduce the minimum to 5 pins is necesary to operate this chip.

[image]

If I put 5VDC on the back light (and his ground). The back light is giving light. If I put 5VDC on the other pin (with the approriated ground to PIN13 of the chip), it doesn't make display something by default.

But something is sure, No pins on the board edge is using COM0-COM3 pins on the chip. What is linked to the chip

3. A5V (Board) -> 100 ohms resistor --> VDD (TM1621)
4. GND (Board) -> GND (TM1621)
5. NA
6. SPI_CS  (Board) -> CSN (TM1621)
7. SPI_DATA (Board) -> DATA (TM1621)
8. SPI_CLK (Board) -> WRN (TM1621)

maxint-rd, 2024-11-25:
> It not seem to work for me... I don't know if due to TM1621 and TM1621D is not the same chip... or something elese like may be my TM1621 is also defective :(

That's too bad...  Have you already tried to power the original MCU to see if the LCD shows anything?
If it does, my next step would be to use a logic analyzer to capture the communication between LCD and TM1621. Some years ago I used that method to reverse-engineer the unknown LCD-matrix display in a bank-card reader. (See my YT video's about the E.dentifier 2).

Most TM16xx chips I have are quite robust and so far I haven't killed even one, regardless the mistakes I made. I would probably also try to see if the LCD segments still work. If I remember well, the YT video also showed some early experiments of applying low voltages to the LCD to test if the segments do react. 

A first comparison of the datasheets of TM1621 vs TM1621D showed that many pins are similar. Both chips have DATA, /WR and /CS. The TM1621 has additional segment pins and also some additional control pins (/RD, BZ and /BZ, OSCO-OSCI, /IRQ) From your research we can conclude that these are all optional and the three pins that matter are the same: DATA, /WR and /CS. For both chips CS is low to activate the chip. When /WR is low data can be written and it is latched when /WR goes high. Voltage levels and operating frequencies seem similar. Display RAM is smaller on the TM1621D, but seems organized the same. 

The commandset of the TM1621D is a subset of the TM1621 command (eg. no RD, OSC, TONE, WDT, etc. on TM1621D). While much different from the TM16xx LED drivers, it seems that both TM1621 chips are fairly similar. My cautious first conclusion is that TM1621D code should also work on TM1621. 

Note however that your LCD panel may require very different settings than the LCD panel used by MakersFunDuck. If you cannot capture the original initialization, I think I would just try different BIAS settings to discover anything that works. (Refer to the TM1621D datasheet for details).

When I find some spare time I want to do my own experiments. I hope you have patience to follow my postings as I cannot promise when that will be. I will keep this issue open and post my progress/findings here. Hopefully together we can make some progress and gain control over our LCD panels...

JGG-Apps, 2024-11-27:
If I put voltage directly on LCD pins, 5V on one segment and ground on one COM, it display something on the LCD, then the LCD himself is not defective.

I need to test the TM1621 himself... find something most basic to generate a valid signal on the output. I have an oscilloscope, I can take a look on the signal... but I'm not totally sure on the signal sync WRN, Clock, etc

Obtenir Outlook pour Android<https://aka.ms/AAb9ysg>

maxint-rd, 2024-11-27:
That LCD experiment is fine. Just don't do it too often or for long times as LCD segments want oscillating values.

I think you should focus first on these two lines: DATA and /WR (or WRN). DATA contains the bits representing data/commands. /WR is actually the clock signal for clocking in the data bits. For each bit written, the /WR line goes high and then low. /CS goes low when communication is starting and may go high to end communication. 

Too bad the original radio MCU is gone. It would have let you see the shape of the signals. But no worry, there's enough continue with.

For these things a logical analyzer is better, but an oscilliscope can be useful too. You could use your oscilliscope to monitor the DATA and /WR signals generated by the MakersFunDuck sketch (while connected to the display). 

If you have more channels on your scope you could also inspect /CS. Since communication is digital, the lines should show blockwave signals, such as shown on page 10 of the TM1621 datasheet (Note that page 3/4 of the TM1621D data sheet shows simpler graphs as that chip only has 3 lines).

First thing to check with your scope is the shape of the signals. They should be fairly rectangular. If not, you may need to add 10K pull-ups to DATA, /WR and /CS. 

You can also check the signal level to see if it gets up to 5V.
Then you can check for noise in the signal. Iffy breadboard connections and long jumpwires can really mess things up. Capacitors can help, but can cause failures at higher frequencies.

Finally you can look at the clock frequency. I don't read chinese, but I think the datasheet mentions a maximum of 300 KHz for 3V and 500 KHz for 5V.
What processor are you using to test? Regular UNO?

maxint-rd, 2024-11-28:
Just a side note: for the TM1621 to work with TM1621D code, there probably should be a pull-up on /RD line. Since the edge connector doesn't even have that line (only /CS, DATA and CLK), I suspect that such pull-up is already somewhere on the board. Based on your picture of the board, it could be R104/R305.

I've also looked at the MakersFunDuck TM1621D.cpp code. Interesting to see it has a TM1621D::readBits() function. Since the TM1621D doesn't even have a /RD pin, I first suspected that this code was written for the TM1621. (In [TM1621D.h](https://github.com/MakersFunDuck/Common-Kitchen-Timer-With-LCD-Replacement-Board/blob/main/Firmware/Print/TM1621D.h) you can read in the comments that the chip also supports external oscillator, just like the TM1621).

However, I also see that the class assumes a shared _RW pin and line 54 says "A class for dealing with the Holtek TM1621D chip.". (Note that TM1621D is by Titan Micro). Looks like the original code was for the HT1621 by Holtek. If you follow the link to the [website mentioned](https://macduino.blogspot.com/), this is confirmed. 

The original code can be found [here](https://macduino.blogspot.com/2015/02/HT1621.html), with English explanation and example sketch. The original HT1621.cpp is unfortunately stored behind a Google drive login.

You could experiment with the simple example there (copied below, just replace "HT1621" by "TM1621D") and perhaps try different BIAS_xxx_n_COM variations to find a match for your LCD panel. As soon as I have  found time to set up my own experiments, I will do the same and let you know my findings.

```
/**
 * \file HT1621-test.ino
 * \brief Simple test sketch for the HT1621 class.
 * \author Enrico Formenti
 * \date 4 february 2015
 * \version 1.0
 * \copyright BSD license, check the License page on the blog for more information. All this text must be
 *  included in any redistribution.
 *  <br><br>
 *  See macduino.blogspot.com for more details.
 */

#include "HT1621.h"

// refere to Macduino website for pin connections and their meaning
HT1621 ht(2,3,4);

void setup() {

  register uint8_t i;
  
  ht.begin();
  
  ht.sendCommand(HT1621::RC256K);
  ht.sendCommand(HT1621::BIAS_THIRD_4_COM);
  ht.sendCommand(HT1621::SYS_EN);
  ht.sendCommand(HT1621::LCD_ON);

  // clear memory
  for(i=0; i<16; i++)
    ht.write(i,0);

  // write all zeroes
  for(i=0; i<16; i++)  
    ht.write(i, 0xBE);
}

void loop() {}
```

maxint-rd, 2025-02-02:
Small update: yesterday I made an English translation of the TM1621D datasheet (v1.0): 
 - [TM1621D_v1.0_Translated_EN.pdf](https://github.com/user-attachments/files/18632006/TM1621D_v1.0_Translated_EN.pdf)

I've soldered a TM1621D chip to a breakout board and hope to find an old LCD suitable for testing soon.
To be continued...

maxint-rd, 2025-03-01:
Another small update: I found a trashed device containing a simple LCD. Connected it to the TM1621D and a CH32V003. 

First attempts to make a TM16xx compatible library gave some interesting findings. The TM1621D is the most simple chip within the TM1621 range. The most complex member has features that are optional to use (such as external clock, readable memory and a timer for beeping a buzzer). Communication follows a known TM16xx protocol but the commands for initializing the LCD are different from those needed to initialize a LED display

The experiments showed that the memory organisation (32 x 4-bits for TM1621) will require some design choices to be made. The 4-bits represent four COM lines. The various TM1621x chips can drive those with up to 32 SEG lines. Page 9 of the [translated TM1621D datasheet](https://github.com/user-attachments/files/18632006/TM1621D_v1.0_Translated_EN.pdf) shows a 7-segment LCD that has each digit use all four COM lines and two SEG lines. Using such connections each single digit can be updated by writing one byte (two half-byte 4-bit nibbles) to the chip. This is most likely not a standardized setup for all LCD displays. So I'm contemplating what facilities I need to implement in the library to provide a generic way to support various different LCD displays.

-----

Then some questions @JGG-Apps : can I ask you some details about your display followed by doing some experimenting with my code?

Your display has 8 digits of 14 segments plus 14 individual segments. Looking at your photo it seems that the 4 COM lines come from underneath the chip, to go to the 4 right-most pins of the LCD connection. _IRQ, BZ, _BZ, OSCO, OSCI, _RD seem not connected. All 32 SEG lines seem to be connected. I can't see all LCD pins; are there 32+4=36 pins in total?

9x14=126; 4x32=128 so that can work. First question is how many COM lines are used per digit. My first attempt for a sketch to find this out would be a test that drives all COM lines one by one, one SEG line at a time. There are some variations possible for the waveform, so we'd first need to find out which initialization works best...

closed 2025-04-08 by maxint-rd