maxint-rd/TM16xx #61: Only one device works at a time

akastormseeker, 2025-03-25:
When using multiple devices (TM1650 in my case), only the first device used works.

My setup:
ESP32C3 micro
4x TM1650 + 4-digit 7-seg displays
Pin D6 connects to all TM1650 data pins
Pins D7-D10 connected to individual clock pins

Code:
```
TM1650 disp1(D6, D7);
TM1650 disp2(D6, D8);
TM1650 disp3(D6, D9);
TM1650 disp4(D6, D10);

void setup() {
  disp1.begin();
  disp2.begin();
  disp3.begin();
  disp4.begin();
  
  for(int digit = 0; digit < 4; digit++) {
    disp1.setSegments(0xff, digit);
    disp2.setSegments(0xff, digit);
    disp3.setSegments(0xff, digit);
    disp4.setSegments(0xff, digit);
  }
}

void loop() {
}
```

Expected behavior:
All displays light up

Actual behavior:
Only disp1 lights up

If I comment out disp1.begin();, then only disp2 lights up

maxint-rd, 2025-03-25:
Hello @akastormseeker , thank you for reporting this. I encountered the very same issue myself when testin two TM1637 displays in Wokwi. I think I found the cause and intend to publish a new release. I wanted to do some physical testing first. Would you perhaps want to assist and test the new code on  your TM1650 project? I will publish new code in the master branch shortly and let you know here...

akastormseeker, 2025-03-25:
I actually think I figured out the problem. When testing, if you just recompile after commenting out #1, then they both will work because the first is already initialized/turned on. The problem is the static "fBeginDone" variable in TM16xx:begin(), which applies to ALL instances of the class. I did a test and made it an instance variable and it works perfectly. I can do a pull request this morning if you'd like.

maxint-rd, 2025-03-25:
Hello @akastormseeker, you're absolutely correct. When I encountered the issue, I first thought it was Wokwi specific, After debugging I found that the static variable was indeed the issue. 

No need for a PR. I already published the fix and published it as a [new release (v0.7.2503)](https://github.com/maxint-rd/TM16xx/releases/tag/v0.7.2503). It may take a day or so before the new release is picked up by the Arduino IDE, but then you should be notified of the updated release. 

Thanks again for submitting this issue with such a clear description. It raised my awareness and made me reconsider the importance of fixing this on short nodice.