maxint-rd/TM16xx #62: TM1640 with generic display class

goephs, 2025-04-14:
I would like to use the additional display features beyond the basic TM1640 class, like printing floating point numbers and not having to figure out where the decimal point should be placed.  I have tried the following simple code:

`#include <TM1640.h>
#include <TM16xxIC.h>

TM16xxIC module(IC_TM1640, 9, 10);   // IC is TM1640, DIO=9, CLK=10
TM16xxDisplay display(&module, 8);    // TM16xx object, 8 digits
void setup() {
  display.println(F("HELLO HI"));
}

void loop() {
  // put your main code here, to run repeatedly:

}`

but I receive a compilation error using an arduino uno and ide 2.3.6 and 2x4 8 digit displays.

> Compilation error: 'TM16xxDisplay' does not name a type

Sorry if I'm missing something basic?  The basic method works fine.

maxint-rd, 2025-04-14:
Hello @goephs, interesting. From first look you're not doing anything special. 
I just gave your a go and got the same error, so I'm probably skipping over the issue. Let me have a look...
(be back in a moment)

maxint-rd, 2025-04-14:
Ah... found it. Your example was missing the include for TM16xxDisplay.h. This one compiles:
```
#include <TM16xxIC.h>
#include <TM16xxDisplay.h>

TM16xxIC module(IC_TM1640, 9, 10); // IC is TM1640, DIO=9, CLK=10
TM16xxDisplay display(&module, 8); // TM16xx object, 8 digits

void setup()
{
display.println(F("HELLO HI"));
}

void loop() {
// put your main code here, to run repeatedly:
}
```

maxint-rd, 2025-04-14:
BTW: you should also be able to use the regular TM1640 class:

```
#include <TM1640.h>
#include <TM16xxDisplay.h>

TM1640 module(9, 10); // IC is TM1640, DIO=9, CLK=10
TM16xxDisplay display(&module, 8); // TM16xx object, 8 digits

void setup()
{
display.println(F("HELLO HI"));
}

void loop() {
// put your main code here, to run repeatedly:
}
```

goephs, 2025-04-14:
Bingo! It works as it should. Many thanks for the solution.  Happy coding ahead.

closed 2025-04-14 by goephs

maxint-rd, 2025-04-14:
Nice! Thanks for letting me know.

The TM1640 is an affordable chip that's great for many different LED projects. I've used them for 16x8 LED matrices of various shapes, which I also combined to make a nice 24x16 LED matrix for the animated DOT-clock (see my video on YouTube). 

Recently I also used one to make an 8-digit 14-segment (plus dot) alphanumeric display. 
16-digit 7-segments plus dot should also work.

Have fun with the TM1640!