maxint-rd/TM16xx #63: Does this library support 16-digit display when using tm1640 to drive common anode diode?

411956158, 2025-04-26:
I used this header file by learning routines.
#include <TM1640Anode.h>

And created a display object.
#define NUM_DIGITS 16
TM1640Anode module1(LED_DIN, LED_SCLK, NUM_DIGITS, true, 3);

Use this command to test.
module1.setDisplayToString("HALLO.111.1111111");

The program is running normally, but only half of it is normal. The first 8 digits of the digital tube are lit normally and HALLO.111 is displayed, while the last 8 digits are not displayed normally, only a few horizontal lines are lit. I have tried many methods but I can't solve them. Do you know why?

maxint-rd, 2025-04-26:
Hello,  that class was made for CA alpha-numeric displays.
The header states "TM1640 with Common Anode, up to 8 digits of 16-segments".
What you're seeing is that the alpha-numeric font is used.
I've tested that class only with 15-segment alpha-numeric CA displays.

However, there is an alternative in which your configuration should be supported, although I've not tested it yet. Maybe you can test the generic TM16xxICanode class? You declare your module object like this:
```
#include <TM16xxICanode.h>
...
TM16xxICannode module1(IC_TM1640, LED_DIN, LED_SCLK, LED_DIN, NUM_DIGITS);
```

When I find some time I will test it myself to see how it behaves for me in such a setup and let you know.

411956158, 2025-04-27:
Thank you for your guidance. I made some attempts.
```
#include <TM16xxICanode.h>
#define NUM_DIGITS 16
TM16xxICanode module1(IC_TM1640, LED_DIN, LED_SCLK, LED_DIN, NUM_DIGITS);

void setup() {
……
  module1.setupDisplay(true, 3);
  module1.setDisplayToString("HALLO.111.1111111");
  delay(5000);
  module1.clearDisplay();
……
}
```
But only 8 digits are still lit. It is gratifying that the horizontal line that the last eight bits will jump with the first eight bits can now be extinguished correctly.

BTW, LED_DIN was used twice in the code you replied to me. I checked the header file, and the second time seems to correspond to strobePin. Is this a clerical error? TM1640 does not have an STB.

[image]

411956158, 2025-04-27:
I tried to test from a lower level, but unfortunately the symptoms didn't change.

```
uint8_t charMap[128] = {0};

#include <TM16xxICanode.h>

#define NUM_DIGITS 16
TM16xxICanode module1(IC_TM1640, LED_DIN, LED_SCLK, LED_DIN, NUM_DIGITS); 

void setup() {

    charMap['0'] = 0b00111111;
    charMap['1'] = 0b00000110;
    charMap['2'] = 0b01011011;
    charMap['3'] = 0b01001111;
    charMap['4'] = 0b01100110;
    charMap['5'] = 0b01101101;
    charMap['6'] = 0b01111101;
    charMap['7'] = 0b00000111;
    charMap['8'] = 0b01111111;
    charMap['9'] = 0b01101111;
    charMap['A'] = 0b01110111;
    charMap['B'] = 0b01111100;
    charMap['C'] = 0b00111001;
    charMap['D'] = 0b01011110;
    charMap['E'] = 0b01111001;
    charMap['F'] = 0b01110001;
    charMap['H'] = 0b01110110;
    charMap['L'] = 0b00111000;
    charMap['O'] = 0b00111111;
    charMap['.'] = 0b10000000;

  module1.setupDisplay(true, 3);
  //module1.setDisplayToString("HALLO.111.1111111");
  const char* str = "HALLO.111.1111111";
  for (byte pos = 0, i = 0; pos < 16 && str[i] != '\0'; pos++, i++) {
    uint8_t segs = charMap[(uint8_t)str[i]];

    if (str[i+1] == '.') {
        segs |= 0b10000000; 
        i++; 
    }
    module1.setSegments16(segs, pos);
  }  
  delay(5000);
  //module1.clearDisplay();

}

```

maxint-rd, 2025-04-27:
Thank you for your efforts to try things and for your elaborate reply. I haven't yet had time to wire a breadboard with TM1640 and CA LED displays. Sorry that I never tested your configuration, but if you can help test things, I'm sure we'll get it to work.

Could you perhaps try the [setSegmentsDebug example](https://github.com/maxint-rd/TM16xx/tree/master/examples/TM16xx_setSegmentsDebug)? Please replace the setSegments call to this:
   `module.setSegments16(nVal, nPos);   // note: you can use setSegments16() when module class supports > 8 segments `
Then experiment with the for-loop values to see what the segments do.

> BTW, LED_DIN was used twice in the code you replied to me. I checked the header file, and the second time seems to correspond to strobePin. Is this a clerical error? TM1640 does not have an STB.
It was no error. There is a constructor for TM16xxICanode with only data and clock pins, but that constructor has no parameter for number of digits. The rest of the code will ignore that strobe setting.

maxint-rd, 2025-04-27:
By the way: what display module is it that you are using? I don't have that module in my collection yet, but I can wire up a bare chip with some CA LED displays to mimic your module.

411956158, 2025-04-27:
> Thank you for your efforts to try things and for your elaborate reply. I haven't yet had time to wire a breadboard with TM1640 and CA LED displays. Sorry that I never tested your configuration, but [...]

thank you again. 
I used that example to do some new tests:
First, I must change the header file to TM1640Anode.h, otherwise the display is garbled; 
Secondly, I did the following tests: 
1. If I use the default for loop, I can change it to nval < = 0xFFFF, so that all the digital tubes will flash in turn (including decimal points); 
2. If I use the annotated for loop and modify it to FFFF, the program will crash after lighting the leftmost two 8' s and decimal points. 

[image]

It will not crash if I modify it to FFFE, but the decimal points of the last eight digits will not light up.
[image]

I hope these will help solve this problem.

maxint-rd, 2025-04-27:
Okay, what display module is it that you are using? I don't have that module in my collection yet, but I can wire up a bare chip with some CA LED displays to mimic your module. Did you buy it somewhere or designed it yourself? Do you have a schematic for me to replicate the connections?

maxint-rd, 2025-04-27:
Just for some quick testing I tried the setSegments example on my 8-digit 15-segment display module to see if it crashes. It didn't. What do you mean  by "crash"? Does it hang? It also didn't garble the display when using TM16xxICanode instead of TM1640Anode.

I have to  figure out what your configuration is. If you have a schematic that would be useful.
After some thinking I'm guessing that your digits may be connected as two sets. The first 8 digits would then be on GRID1-GRID8 and the second 8 digits to GRID9-GRID16, with both sets sharing the SEG1-8 lines as common anodes to specify the digit. In issue #53 someone asked about a similar setup. Can you please tell me how the segments are connected? Ar the SEG lines shared?

BTW. I see you're using an Arduino nano. On 8-bit processors the int is only 16-bits. The <= may have a wrapping issue. You may want to change the inner for-loop to use a 32-bit integer:
```
    for(uint32_t nVal=1; nVal<=0xFFFF; nVal=nVal*2)
    {
      module.setSegments16(nVal, nPos);   // note: you can use setSegments16() when module class supports > 8 segments (TM1638, TM1668)
      delay(100);
    }
```

A possible more readable variation would be 
```
    for(byte nSeg=1; nSeg<=16; nSeg++)
    {
      module.setSegments16(bit(nSeg), nPos);   // note: you can use setSegments16() when module class supports > 8 segments
      delay(100);
    }
```

I've tested both variations with both TM16xxICanode and TM1640Anode and saw no crashes or hanging. Next step is to wait for your response to see how to connect 8-segment digits to a bare chip.

411956158, 2025-04-27:
Sorry, I forgot to upload it.[image]

411956158, 2025-04-27:
> Sorry, I forgot to upload it.[image]

the mcu is esp32c3.ESP32-C3 integrates a 32-bit core RISC-V microcontroller with a maximum clock speed of 160 MHz.

maxint-rd, 2025-04-27:
Great! Thanks for the schematic. This does make it clearer and confirms my assumption about the connections. It is like the configuration in issue #53. At the moment such configuration is not supported, but with help of your testing, I am willing to make an addition of TM1640Anode to support that configuration.

With the current TM1640Anode class the altered setSegments example should light up all segments of all digits in the order of top-1, bottom-1, top-2, bottom2, etc., But to function as intended you need to `#define NUM_DIGITS 8 `. I suggest you use this inner for-loop:
```
    for(byte nSeg=1; nSeg<=16; nSeg++)
    {
      module.setSegments16(bit(nSeg), nPos);   // note: you can use setSegments16() when module class supports > 8 segments
      delay(100);
    }
```

Regarding the crashing: do you see some message of the ESP resetting or something else? I have a feeling this may be an ESP32 specific issue related to power usage. Many months ago I had a TM1640 on a PICO becoming unstable when more segments were lit.   After quite some time my conclusion was that the specific chip was more sensitive than others for power issues. See [this comment](https://github.com/maxint-rd/TM16xx/issues/34#issuecomment-2125623415) of issue #34 for details.

maxint-rd, 2025-04-28:
FYI: I'm working on a new class to support this 2 x 8-digit x 8-GRD  CA 7-segment configuration. Main difference with the 8-digit x 16-GRD CA configuration is the layout of the memory buffer used to transpose segments and digits to match the display memory of the chip. Both the current TM1640Anode class and the the TM16xxICanode class treat that memory as one block, but for this configuration it needs to be split into two 8x8 blocks. To be continued...

maxint-rd, 2025-04-29:
Update: I just published a first version of the TM1640Anode8S16D class to support the 2 x 8-SEG x 8-digit Common Anode configuration on TM1640/TM1640B.

I've tested it with a bare TM1640 chip, CA LED display with CH32X033 MCU, but ESP32 should work as well.
Just include TM1640Anode8S16D.h instead of TM1640Anode.h. 
(The somewhat cryptic 8S16D stands for 8-segments x 16-digits).
As mentioned above the main difference with TM1640Anode is the handling of the offscreen memory bitmap.

I'd appreciate if you could test it and let me know your findings.

411956158, 2025-04-29:
Thank you very much for your efforts. I will try it as soon as possible and let you know as soon as I have the result.

411956158, 2025-05-01:
Thanks to your work, now he can use "print" to work correctly, but when using "setDisplayToString", the first character is not displayed (whether it is a number or a character).（"TM16xxDisplay.h" was not include when testing "setDisplayToString".）

```
#define LED_SCLK     12   
#define LED_DIN      13   

#include <TM1640Anode8S16D.h>
#include <TM16xxDisplay.h>

#define NUM_DIGITS 16
TM1640Anode8S16D module1(LED_DIN, LED_SCLK, NUM_DIGITS, true, 3); 
TM16xxDisplay display(&module1, NUM_DIGITS);    // TM16xx object, 8 digits

void setup() {
  module1.setDisplayToString("HALLO.111.1111");
  delay(5000);
  module1.clearDisplay();
  display.println(F("HELLO11.123.456.789"));
}
```

module1.setDisplayToString("HALLO.111.1111");

[image]

display.println(F("HELLO11.123.456.789"));

[image]

411956158, 2025-05-01:
Sorry, I accidentally clicked "comment and close"

maxint-rd, 2025-05-01:
Okay, seems we're heading in the right direction, but not quite there. I'll reopen the issue and try to reproduce your problem.

maxint-rd, 2025-05-01:
Hmmm. Interesting... Good find!
I was able to reproduce it (which is good news), but I'm now doing some real-time debugging to find the cause. In the mean time I can give you a work-around: 

The display state and intensity parameters are deprecated and no longer used by the constructor. You can declare your module object like so:
`TM1640Anode8S16D module1(LED_DIN, LED_SCLK, NUM_DIGITS); `

To fix issues with timing-interrupts on some processors, these parameters were moved to `begin()`, which you can call at the top of setup:
```
void setup() {
   module1.begin(true, 3);
   module1.setDisplayToString("HALLO.111.1111");
   ....
}
```

Note that calling `begin()` explicitly should be optional. If you don't call it, the first sending of data triggers an implicit call. To set display state and intensity you then call `setupDisplay()`.
[Going back to debugging now. To be continued...]

maxint-rd, 2025-05-01:
Update: found the cause: the implicit `begin()` cleared the memory bitmap while filling it. 

Just now I [published a fix](https://github.com/maxint-rd/TM16xx/commit/b4e5bd577f5315d2afe741914c98abcbbb1c04c0). Please update your local version of TM1640Anode8S16D.cpp. After updating that file, your original example (without explicit `begin()`) should work just as well, (Although you still need to use `begin()` or `setupDisplay()` to set display state and intensity).

maxint-rd, 2025-05-12:
Hello @411956158, I think I can close this issue since everything should be working now. Please go ahead and post your findings in this issue. I will reopen it when needed, Thank you for your effort and time to help me implement support for this configuration.

closed 2025-05-12 by maxint-rd

411956158, 2025-05-12:
Sorry, recent business trips and meetings have taken up all my time. Once I have some free time, I will do more tests on him and share my findings with you.

maxint-rd, 2025-05-12:
I understand (Work goes before pleasure). Thanks for letting me know.