maxint-rd/TM16xx #64: Using I2C Wire library with TM1650

vanniaz, 2025-05-06:
The README file says:

> The TM1650 datasheet mentions SDA and SCL pins. The used protocol resembles I2C, but lacks addressing. For that reason this library doesn't use the I2C Wire library, but (slow) bitbanging using digitalWrite.

However, this repository [https://github.com/arkhipenko/TM1650](url) shows that I2C **can** be used. If I understand it well, the command code can simply be used as I2C address. Thus bitbanging could be avoided.

Could Wire library be used also in TM16xx?

maxint-rd, 2025-05-07:
Hello, thank you for bringing this up (and for actually reading the readme!)
Using true I2C directly on the TM1650 sounds interesting. I haven't read much about others doing that, so thank you for that link. I think offloading communication to hardware can be usefull and is worth investigating.

Just for your information: in other areas I did use I2C:
 - Currently I'm working on the TM1680. With 24-seg x 16-com it's like a larger version of the TM1640. It's similar to the TM1681, but uses true I2C (addresses 0x72/0x73).
 - Earlier this year I added support for the [QYF-0231 14-segment display](https://github.com/maxint-rd/TM16xx/commit/1433d9e267dd8aeb47e2d256942e882c35fc2e2f), a 4 digit 15-segment alpha-numeric LED module that uses the HT16K33 chip. That chip resembles the TM1640, but uses I2C (default address 0x70).
 - Last year I published the [TinySoftWire library](https://github.com/maxint-rd/TinySoftWire) to interface with peripheral devices such as sensors using software I2C on ATtiny MCUs. It even includes an example for a [TM1650 I2C gateway](https://github.com/maxint-rd/TinySoftWire/tree/main/examples/I2C_gateway_TM1650). Using that library I could use multiple of [these TM1650 modules](https://github.com/maxint-rd/arduino-modules/wiki/09-TM1650-4x7-segment-display) at different addresses.

For that last example I already converted the TM1650 class to use I2C. Wonder how that works when used directly. If I understanding your post well, I would need to make some changes with regard to sending commands/data. Would it mean different addresses for changing brightness, reading buttons, etc.? Sounds like an interesting experiment!

maxint-rd, 2025-05-07:
When adding your link to the [links section](README.md#links) of my README, I saw that I earlier found someone else who made a [I2C-based TM1650 driver](https://github.com/mozgy/Mozz_TM1650).

When comparing the two I saw they both use the same approach of using ranges of I2C addresses to send data/commands. Both were made in 2015 and while the code is different they do have design similarity and constants named the same, so one may be based on the other. Both do not support button key scanning, although [some people tried](https://github.com/arkhipenko/TM1650/issues/1) to add that. I also saw a bug regarding the number of digits, but then read the author never saw a TM1650 datasheet.

What I found interesting is that both libraries assume that brightness can be set for each digit individually. Even after another close read of the datasheet (v2.1 EN), I could not find such feature. I guess that's good for another experiment.

Since it kind-of defeats the addressable nature of I2C, I'm not too fond of using a range of I2C addresses. Using hardware I2C could be useful to achieve higher speeds, although the datasheet states "The recommended value is below 100kHz."
What do you think would be other advantages of using I2C instead of bit-banging?

vanniaz, 2025-05-07:
I think that both approaches (bit-banging vs range of I2C addresses) have pros and cons. In my case I am designing a board where the only I2C peripheral is the TM1650, and since the microcontroller is used for audio signal processing, I try to keep the CPU load for the display as low as possible. In other cases it could be better to accept a somewhat higher CPU load to save I2C addresses.
I will try the Wire based code on my prototype, and I will let you know if it works as expected.
I had not noticed the "below 100 kHz" suggestion, this is below the standard I2C rate and I will need to call setClock(), thanks for pointing out!

maxint-rd, 2025-05-07:
Okay. In the past I've done some measurements using a logic analyser, but I can't quite remember what the achieved speeds were on the various microprocessors. 

Years ago, when testing on the ESP8266 I noticed that reading button state required lower speeds on most TM16xx chips, So for fast processors a 5 microsecond bit-delay was introduced.

I think 100 kHz will work fine, so just give it a go and adjust when needed.

BTW: bitbangjng I2C on the ATtiny13A running at 9.6MHz allowed reaching standard 100 kHz I2C speed.
What microprocessor are you using at what clock-frequency?

vanniaz, 2025-05-07:
I am using an ESP32 (240 MHz) which of course should be capable of bit-banging at very high frequency, but I am using it as a DSP and I want to save every possible CPU cycle for the audio processing, so I will go for the I2C. I will probably set the I2C speed well below 100 kHz, because I want to stay on the safe side (and a few milliseconds to update a 7-segment display are not an issue)

maxint-rd, 2025-05-07:
Clear. I'm not very experienced with the ESP32, but I know it has some other useful features. Using seperate cores can offload the cores used for DSP and best of course would be to use hardware communication with DMA and perhaps an interrupt routine to maintain state. One of the things I still want to try is driving a TM16xx chip using hardware SPI. (So many things, so little time...)
Good luck in your experiments! (I appreciate you letting me know your findings...)

vanniaz, 2025-05-09:
I tried the I2C code (from the library you mentioned [https://github.com/mozgy/Mozz_TM1650](url)) and it seems to work reliably.

I also tried increasing the I2C speed to 400 kHz and it did not show any problem. However the difference in display update time is barely noticeable, so I will stay at 100 kHz or less.

I will not use this library as it is, because some aspects don't convince me (e.g. keeping an image of the control register for each individual digit, which is probably meaningless). As you already noticed, the code seems to assume that it is possible to set a different brightness level for each digit, but then an overload of the SetBrightness method writes to digit 0 to set the entire display, which proves that there is only one control register (unless I'm overlooking something).

So I will keep the I2C communication but I will use parts of your library, which is more complete.
Thanks for your support

maxint-rd, 2025-05-09:
Good to know that the Mozz library works. In theory using DMA could off-load the ESP32, but I just read in some topics that setting up DMA may take too long and might even not be supported (yet) in the Arduino core. (Don't know the latest state though). 

BTW: My remark about I2C in the readme originates from the v1.0 datasheet that labels the interface pins as SDA and SCL and also how ACK is used similar to I2C, while not including an address. I noticed that in the latest datasheet v2.1 these pins are now labeled CLK and DAT; more in line with the other TM16xx chips.

With regard to the Mozz library code: I also find that TM1650_CONTROL_STORE meaningless. As far as I can see there is no individual digit manipulation, other than the segment memory. Maybe the code was based on some other library that allowed digit specific settings, but all the LED driver chips I know have things like brightness and standby state as global setting.  (I just double checked two other popular chips: HT16K33 and MAX7219). 
I'm not too fond of the TM1650_DIGITS_STORE either, but it is used by `SetDot()` to switch off/on the dot separately from the number. (My library follows the approach used in the original TM1638 library, in which the dot is always set together with the number). Further more: The `ColonON()`/`ColonOFF()` functions seem out of place (using the control address). The use of "segment" in some functions, instead of "position" is perhaps a bit confusing.

Feel free to use any library code you like. I'll probably close this issue, but in case you like to share your work, you can always drop a link here. Good luck with your project!

vanniaz, 2025-05-09:
Yes, I completely agree that also TM1650_DIGITS_STORE is not useful, I will follow your method of setting the dot together with the number, if required, and that "segment" should be renamed to "digit" (or "position").

I started from the original code just to check how it worked, but once I know that I2C is reliable I will rewrite it.

I'll probably tailor my code to my application, so I am not sure whether it will be interesting to others, however I'll see what comes out.
I think we have discussed every detail so we may close this issue, thank you again!

closed 2025-05-09 by maxint-rd