maxint-rd/TM16xx #65: TM1623 only 4 column

olivierpls, 2025-05-18:
Hello, 
I'm having a little trouble using this module with the TM16xx library.

In my code, I included "tm1623.h" to correspond to my module, but I can't get all the segments to work, only the first four columns work.
I reassembled the module in the device I took it from, and all the LEDs work fine, but I can't get them to work with the Arduino. 

```cpp
#include <TM1623.h>
#include <TM16xxMatrix.h>

TM1623 module(8, 9, 7);
#define MATRIX_NUMCOLUMNS 7
#define MATRIX_NUMROWS 7
TM16xxMatrix matrix(&module, MATRIX_NUMCOLUMNS, MATRIX_NUMROWS);

void setup()
{
  matrix.setAll(true);
  delay(200);
  module.setupDisplay(true, 1);
  delay(400);
  matrix.setAll(false);

}

void loop()
{
  matrix.setAll(true);
  delay(200);
  module.setupDisplay(true, 1);
  delay(400);
  matrix.setAll(false);
  delay(1000);
}
```

If you could help me, thank you!

maxint-rd, 2025-05-18:
Hello @olivierpls , interesting board. Thank you for posting that image. I wasn't aware of its existence yet. I see VTech printed at the bottom. Did you salvage this module from some toy? Can you tell a bit more about it?

The TM1623 has different modes for configurable segments x grid lines: 11 x 7 - 14 x 4 . The constructor you're using has a default of 4 digits, ie. 4 common grid lines (and 14-seg lines). In your picture it seems SEG8-SEG11 lines are unused and pins 22-24 are used as GRID5-GRID7. By specifying a different number of digits in the TM1623 constructor a different mode can be selected. (I tested the TM1623 class with an alpha-numeric display, not with a matrix). 

You can try this constructor instead:
`   TM1623(byte dataPin, byte clockPin, byte strobePin, byte numDigits);`

So in your case that would become this:
`    TM1623 module(8, 9, 7, 7); // data=8, clk=9, stb=7, numDigits=7`

Could you give that a try? 
If the matrix example still gives issues, could you then also try the [TM16xx_setSegmentsDebug example](https://github.com/maxint-rd/TM16xx/tree/master/examples/TM16xx_setSegmentsDebug) and adjust the loop/constuctor as needed?

olivierpls, 2025-05-19:
First of all, thank you so much for the quick response!

Indeed, I recovered this module from a Vtech educational game for children with the reference "1546" and under the name: "**Touch & Learn Activity Desk**". 
Attached are some images with the LED matrix module removed from the device.

[https://imgur.com/a/lfx5Ysx](https://imgur.com/a/lfx5Ysx)

I tried it with your constructor, and it works great now, thank you!
In the meantime, I found another solution that worked. 

I had to use this constructor: ``` TM1628 module(8, 9, 7); ``` and it worked the first time!

Thanks again for your help :)

maxint-rd, 2025-05-19:
Thank you for giving the module some more context. The readme has a [list of real-world devices](https://github.com/maxint-rd/TM16xx#real-world-devices-using-a-tm16xx-chip). I will add this one with your pictures.

Good to see you got it working and found that TM1628 is very similar to TM1623. The main difference is one segment pin less and only two key pins instead of three. 
In the class implementation there are some other differences, such as the default for number of digits. The reason is that TM1623 is more suited for 14-segment alpha-numeric display with 4 digits, so I chose that as default.

For your matrix using the TM1628  class may be best, because the TM1623 class is derived from the generic TM16xxIC class, giving some overhead in flash usage.

maxint-rd, 2025-05-20:
### TM1623 - **vTech 1546 Touch & Learn Activity Desk**

[image]

[image]

[image]

[image]

closed 2025-05-20 by maxint-rd

maxint-rd, 2025-05-20:
Thank you @olivierpls for your contribution. I added your pictures above and made a reference to the [list of real-world devices](https://github.com/maxint-rd/TM16xx#real-world-evices-using-a-tm16xx-chip). Always nice to see where TM16xx chips have been used.

The TM1623 supports key-scanning of 10 x 3 = 30 buttons, so I perhaps it also handled the 6 x 5 alphabet/animal input keys of the keyboard, with possible direct connection to a microcontroller for the remaining buttons/switches.

olivierpls, 2025-05-20:
Thank you for your help and I'm glad I was able to contribute to the project! 

I think that in my case the TM1623 module was only used for the display. 
The pressing of the keys on the keyboard is managed by microcontrollers embedded in black resin and impossible to identify. 

The program data (sounds, voices, etc.) are stored on an MX25l6406E SPI FLASH memory. 

Attached are pictures of the circuit (the spi flash memory is missing, I have already unsoldered it to extract the contents).. 

I like to try to understand how these interactive circuits work within the limits of my skills lol

[image]
[image]
[image]
[image]

maxint-rd, 2025-05-20:
Very interesting to see. I share your interest and also like to know how things work. 
Thanks for this addition. Feel  free to post in this issue anything else you find out.
Have fun!